micropython/micropython #7486: Warning: micropython.org SSL certificate is not validated

bar2104y, 2021-07-04:
when trying use upip.install fo any package i have this error. How fix it?

domdfcoding, 2021-07-04:
I am encountering the same issue.

Built from master as follows:
```console
cd mpy-cross
make
cd ../ports/unix
MICROPY_PY_USSL=1 make submodules 
MICROPY_PY_USSL=1 make deplibs
MICROPY_PY_USSL=1 make
```

The error is as follows:

```console
$ ./micropython -m upip install <package>
Installing to: /home/domdf/.micropython/lib/
Warning: micropython.org SSL certificate is not validated
Error installing '<package>': , packages may be partially installed
```

domdfcoding, 2021-07-04:
I'm not sure the two issues I'm seeing are related, so I have opened #7487 for the second error.

bar2104y, 2021-07-04:
Yes, this is one mistake.

But I'n trying use console on esp8266:
```
>>> import upip
>>> upip.install("package_name")
```

berkut0, 2022-04-06:
I confirm.
Still the same error, Micropython 1.18 stable release

berkut0, 2022-04-07:
And more interesting that after successful installation of "micropython" package new error appears — " packages may be partially installed"

``` Python
>>> import upip
>>> upip.debug = True
>>> upip.install("micropython-pystone")
Installing to: /lib/
Queue: ['micropython-pystone']
https://micropython.org/pi/micropython-pystone/json
Warning: micropython.org SSL certificate is not validated
Installing micropython-pystone 3.4.2.post2 from https://micropython.org/pi/pystone/pystone-3.4.2.post2.tar.gz
https://micropython.org/pi/pystone/pystone-3.4.2.post2.tar.gz
Skipping micropython_pystone.egg-info/PKG-INFO
Extracting /lib/pystone.py
{}
>>> import gc
>>> gc.collect()
>>> upip.install("python-telegram-bot")
Installing to: /lib/
Queue: ['python-telegram-bot']
https://micropython.org/pi/python-telegram-bot/json
https://pypi.org/pypi/python-telegram-bot/json
Error installing 'python-telegram-bot': , packages may be partially installed
>>> 
```

Jachimo, 2022-10-12:
Any updates on this issue? I am running into it while trying to install `aioble-central` on ESP32.

This is after connecting the board to WiFi with access to the Internet:

```Python
>>> import upip
>>> upip.install('aioble-central')
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'aioble-central': Package not found, packages may be partially installed
>>> 
```

I'm unsure where to go from here, and this issue is one of the top search results, so it seems this just… doesn't work? Or is that package not one that's supported on the ESP32 architecture/port, and the error message is just being unhelpful?

jimmo, 2022-10-12:
@Jachimo You're right, the error message is being unhelpful, and the actual issue is nothing to do with the SSL certificate. That's just an FYI from upip. The "Package not found" is the important bit.

What's going on here is that `upip` has been unmaintained for a while and does not have any of the new packages like `aioble`. We have been working on its replacement `mip`, which will be included in v1.20, and is currently in the nightly builds.

So the simple answer is, if you can update to nightly firmware, then you will be able to do:

```
>>> import mip
>>> mip.install("aioble-central")
```

(The micropython.org server cert will still not be validated, but we're working on that separately).

Alternatively, if you don't want to update the firmware you can also use the mip functionality from `mpremote`. With the latest published version (0.4.0), you can do

```
$ mpremote connect /dev/ttyUSB0 mip install aioble-central
```

Some more notes on upip->mip here: https://github.com/orgs/micropython/discussions/9565#discussioncomment-3837770

jonnor, 2024-10-01:
upip has been replaced by mip in MicroPython 1.20, https://github.com/micropython/micropython/releases/tag/v1.20.0
Closing as no longer relevant.

closed 2024-10-01 by jonnor