micropython/micropython #7487: AssertionError in upip when package has multiple files on PyPI

domdfcoding, 2021-07-04:
I have been attempting to install packages from PyPI using ``upip``, but I am encountering the following error:

```console
$ ./micropython -m upip install <package>
Installing to: /home/domdf/.micropython/lib/
Warning: micropython.org SSL certificate is not validated
Error installing '<package>': , packages may be partially installed
```

I have tracked it back to an assertion on line 201:

https://github.com/micropython/micropython/blob/0e11966ce9ea2345458d2e6127ee28486c4cb65d/tools/upip.py#L194-L202

However, if the project has multiple files (quite common, e.g. sdist and wheel) the assertion is false, as `data["releases"][latest_ver]` contains an entry for each file and therefore the length exceeds 1.

aivarannamaa, 2021-08-22:
Related to #7295

jonnor, 2024-10-01:
upip has been replaced by mip in MicroPython 1.20, https://github.com/micropython/micropython/releases/tag/v1.20.0
Closing as no longer relevant.

closed 2024-10-01 by jonnor