micropython/micropython #7488: Hang when reading from UART and writing to filesystem

murzoh, 2021-07-04:
When reading from UART and trying to write the read line to a file the board has a hard hang. I've confirmed the following code causes the issue on both an RPI Pico and a Pimorini Pico Lipo. The hang occurs anywhere from the first iteration of the loop, most often early, but can run to 10 or 20.

*Note: I'm not sure if this is the same issue as [this one](https://github.com/micropython/micropython/issues/7204) but thought I'd raise anyway.

I've run the same code on an ESP-32 board (Firebeetle v4)  with no issue. The UART is read and line written to file with no issues.

Sample code that draws the error (apologies I'm not the best coder so apologies for the quality of the code):
import os
import machine

#print sys info
print(os.uname())

f_debug = open('debug_test116.log', 'a')
f_debug.write('=======STARTED =======\n')
f_debug.flush()

#print uart info
uart = machine.UART(1, baudrate=9600, txbuf=1024, rxbuf=1024, timeout=1000, timeout_char=4 )
print(uart)

i = 0
while i < 1000:
    line = uart.readline()
    f_debug.write(str(line))
    f_debug.flush()
    print(line)
    i = i + 1
    
print()
print("- bye -")

purpleskyfall, 2023-05-11:
I had the same problem using MicroPython v1.20.0 on RPI Pico, is there any progress?

Gadgetoid, 2026-08-20:
The main issue I had trying to write data on the fly from a UART is that bytes would be dropped (the internal buffer is only 32 bytes) while the flash erase/write cycle ran. The naive choice to use a large ring-buffer and an IRQ to try and catch these fails for the same reason, since the UART-buffer overflows before the IRQ can fire and hand over to the ring-buffer.

Even without a hang, the data loss can only be recovered by using flow control which turned out to be something of a miraculous silver bullet (who knew, eh!).

Otherwise pretty much only DMA from UART into an SRAM ring-buffer can save you on RP2, but there are a whole bunch of caveats that prevent that being the go-to in MicroPython.

Either way I could not repro this hang, so probably fixed? But the above Python code is nonetheless flawed and almost guaranteed to quietly lose bytes when the UART buffer saturates during flash writes. (which is unrelated whatever this specific bug might have been, but is the silent, frustrating trap I was getting at in #17236.)

agatti, 2026-08-20:
I'm not an expert on how the RP2 works at a lower level, but can't this be solved the other way around?  As in, use a dedicated DMA channel to handle flash I/O and then use an IRQ to signal completion to let further writes through.

Gadgetoid, 2026-08-21:
> use a dedicated DMA channel to handle flash I/O

It's... complicated.

Flash is memory-mapped using XIP (eXecute In Place) so that program code doesn't have to care about the implementation details of how and where it's running. *Writing* flash (including the erase step) takes it out of XIP temporarily and into a more conventional bulk IO mode.

This means program code in flash *cannot execute* while flash operations are happening. You can work around this by loading your critical interrupt routines into SRAM, or using an unattended DMA into a ring buffer, but MicroPython has no way of knowing how to unpick your specific use-case and do the right thing.

There are many approaches to keeping UART alive during flash IO that work for specific use-cases, but which break the wider surface area of MicroPython's UART implementation. For example RX/TX IDLE cannot fire if an unattended DMA is continuously servicing the UART buffer. Moving the UART buffer -> ring buffer handler into SRAM costs SRAM (even if you're not using it), and even then you cannot guarantee that the ring buffer itself is in SRAM (it might be in PSRAM), nor guarantee that any interrupts the user might hang off UART aren't otherwise trying to touch flash.

The snarky TLDR is that you cannot DMA into flash, because the m stands for "memory" and flash is not memory in this sense, even if it might be mapped as such for convenience.

The only reasonable fixes are to use UART flow control, or write C code that makes the guarantees MicroPython cannot. Nonetheless I'd love to be proven wrong on this, 'cos it's a real gotcha!

agatti, 2026-08-21:
Thanks for the explanation!

For what it's worth, the STM32 port operates the way I mentioned: having the functions used to writing to flash and the UART ISR kept in RAM - the latter also having a permanent ringbuffer as the STM32 UARTs have no FIFOs.  The only difference being that flash writes do not use DMA but are IRQ based instead.

It's a pity this is not easily achievable on the RP2, as there's even a dedicated XIP interrupt (on the 2040 at least) :|