micropython/micropython #7489: Only GPIO18,GPIO19,GPIO25,GPIO26 can be used for I2C port on ESP32

yangwc, 2021-07-05:
When I use the I2C port of ESP32, only GPIO18,GPIO19,GPIO25,GPIO26 can be configed to I2C port. The other GPIO can't be used by I2C.

dpgeorge, 2021-07-05:
Please show the code that you use and exactly what's not working.  It should be possible to select the SCL/SDA pins via the keyword arguments `scl` and `sda` in the `machine.I2C()` constructor.

yangwc, 2021-07-06:
> Please show the code that you use and exactly what's not working. It should be possible to select the SCL/SDA pins via the keyword arguments `scl` and `sda` in the `machine.I2C()` constructor.

Thanks, when config like this: "i2c = I2C(1, scl=Pin(5), sda=Pin(18), freq=5000)" , it does not work.  
When use "i2c = machine.I2C(scl = machine.Pin(5), sda = machine.Pin(18), freq = 100000)" instead, it's OK, and there is a warning: I2C(-1, ...) is deprecated, use SoftI2C(...) instead .

jimmo, 2021-07-06:
> Thanks, when config like this: "i2c = I2C(1, scl=Pin(5), sda=Pin(18), freq=5000)" , it does not work.

I just ran this exact line on my ESP32 and it worked. What's the issue that you see? (Does it give an error or is it that you can't communicate with your device?)

> When use "i2c = machine.I2C(scl = machine.Pin(5), sda = machine.Pin(18), freq = 100000)" instead, it's OK, and there is a warning: I2C(-1, ...) is deprecated, use SoftI2C(...) instead .

The warning is because you're no longer specifying a hardware I2C peripheral ID, so it's using the software implementation instead. It's better to use SoftI2C instead (to be explicit that this is what you want).

I ntoice that you're using freq=5000 and freq=100000 in the second.

yangwc, 2021-07-08:
> > Thanks, when config like this: "i2c = I2C(1, scl=Pin(5), sda=Pin(18), freq=5000)" , it does not work.
> 
> I just ran this exact line on my ESP32 and it worked. What's the issue that you see? (Does it give an error or is it that you can't communicate with your device?)
Thanks, I have tried "i2c = I2C(1, scl=Pin(5), sda=Pin(18), freq=5000)" one more time, but at this time, it works OK! Maybe before the hardware isn't work correcttly.
> The warning is because you're no longer specifying a hardware I2C peripheral ID, so it's using the software implementation instead. It's better to use SoftI2C instead (to be explicit that this is [...]

closed 2021-07-08 by yangwc