micropython/micropython #7493: STM32: WLAN.deinit(STA_IF) twice causes sdio_transfer timeout

fragmuffin, 2021-07-06:
Deinitialising an WLAN(STA) object twice causes micropython to become unresponsive.

# Stimulus
`WLAN(STA)` object is:
- instantiated
- activated
- de-initialised
- de-initialised (a 2nd time, before activating)

**To reproduce**: run the code:
```python
import network
sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.deinit()
sta_if.deinit()
```

```
MicroPython v1.16-58-g3966f6774 on 2021-07-06; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import network
=== sta_if = network.WLAN(network.STA_IF)
=== sta_if.active(True)
=== sta_if.deinit()
=== sta_if.deinit()
=== 
sdio_transfer timeout STA=0x00000000
sdio_transfer timeout STA=0x00000000
sdio_transfer timeout STA=0x00000000
sdio_transfer timeout STA=0x00000000
sdio_transfer timeout STA=0x00000000
... output truncated
```

### Workaround
Manually keep track of whether the object has been initialised:

```python
from network import WLAN, STA_IF

_sta_if = None

def connect(*args):
    global _sta_if
    if not _sta_if:
        _sta_if = WLAN(STA_IF)
    _sta_if.active(True)
    _sta_if.connect(*args)

def disconnect():
    global _sta_if
    if _sta_if:
        _sta_if.disconnect()
        _sta_if.deinit()
        _sta_if = None
```

```python
>>> connect('my_ssid', 'secret')
>>> # multiple calls survived...
>>> disconnect()
>>> disconnect()
```

fragmuffin, 2021-07-06:
possibly relates to: https://github.com/micropython/micropython/issues/4696

dpgeorge, 2021-08-31:
Fixed by 52a78e69658eca65bcb8d6577df2de2db7932c9e

closed 2021-08-31 by dpgeorge