micropython/micropython #7495: ESP32 - ADC sampling rate much slower with V1.15 than V1.16

pidou46, 2021-07-06:
Hello,

I run the following code with V1.15 and V1.16 firmware and get much less samples with V1.16

```
from machine import Pin, ADC
from micropython import const
import machine
import time
from array import array

#first test show a 35_us ADC polling interval with a ESP32-S module without spiram it's about 28.5_kHz
#I will acquire 600 samples to be shure to get enough sample to get a full 50Hz period (20_ms)
SAMPLES=const(600)

data=array("l", SAMPLES*[0])
timestamps=array("l", SAMPLES*[0])
pin=ADC(Pin(32))

@micropython.viper
def sample(data, timestamps, p, t):
    state=machine.disable_irq()
    for _ in range(SAMPLES):
        data[_]=p.read()
        timestamps[_]=t()
    machine.enable_irq(state)

def run():
    start_run=time.ticks_us()
    sample(data, timestamps, pin,  time.ticks_us)
    sum_sqr=0
    n=SAMPLES-1
    start=timestamps[n]
    #I keep the last 20_ms (1/50) samples: 1 full 50_Hz period
    while start-timestamps[n] < 20000:
        #print('n:{} {}, {}'.format(n,timestamps[n],data[n]))
        sum_sqr+=data[n]**2
        n-=1
    print('samples: {}'.format(SAMPLES-n))
    print('RMS: {}'.format((sum_sqr/(SAMPLES-n))**0.5))
    end_run=time.ticks_us()
    print('running time: {}_µs'.format(end_run-start_run))

#give some extra samples with V1.16
#machine.freq(240000000)

run()
```

Results:

```
MicroPython v1.15 on 2021-04-18; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
samples: 577
RMS: 2.427457
running time: 31764_µs
```

```
MicroPython v1.16 on 2021-07-06; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
samples: 196
RMS: 719.0552
running time: 70425_µs
```

With 240_Mhz

```
>>> %Run -c $EDITOR_CONTENT
samples: 210
RMS: 709.0872
running time: 65361_µs

```

pidou46, 2021-07-06:
CPU is:

ESP32-D0WDQ6 (revision 1)

jimmo, 2021-07-07:
@pidou46 Would it be possible to try and narrow down your test a bit to confirm that it's definitely the ADC.

For example, a simpler program that just does

```python
start_run=time.ticks_us()
for i in range(200):
  p.read()
end_run=time.ticks_us()
print('running time: {}_µs'.format(end_run-start_run))
```

Also a good idea to disable the GC during any performance-critical sections to avoid a collection. Also worth considering locking the GC to ensure that your code doesn't do any allocations. (Although watch out for time.ticks_us which will allocate when the device has been on for more than 2147 seconds).

Note we do see large performance variations between ESP32 releases due to unpredictable code organisation and CPU caching effects. But this does seem fairly dramatic.

pidou46, 2021-07-07:
Here is a result of the test code:

```
MicroPython v1.16 on 2021-06-23; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 11109_µs
```

```
MicroPython v1.15 on 2021-04-18; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 12595_µs
>>> %Run -c $EDITOR_CONTENT
running time: 12581_µs
```

The difference is small and can be explain buy GC.

But the test code is not viper code.

My guess is that difference comes from the viper performance between V1.15 and V1.16

I will try this as soon as possible

pidou46, 2021-07-07:
```
import time
from machine import Pin, ADC

p=ADC(Pin(32))

@micropython.viper
def sample():
    for i in range(200):
      p.read()

start_run=time.ticks_us()
sample()
end_run=time.ticks_us()
print('running time: {}_µs'.format(end_run-start_run))
```

```
MicroPython v1.15 on 2021-04-18; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 8606_µs
```

```
MicroPython v1.16 on 2021-07-06; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 9397_µs
```

Not abovious

pidou46, 2021-07-08:
Another try with cg control

```
import time
from machine import Pin, ADC

p=ADC(Pin(32))

@micropython.viper
def sample():
    for i in range(200):
      p.read()

gc.collect()
gc.disable()
start_run=time.ticks_us()
sample()
end_run=time.ticks_us()
gc.enable()
print('running time: {}_µs'.format(end_run-start_run))
```

```
MicroPython v1.16 on 2021-07-06; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 9348_µs
>>> %Run -c $EDITOR_CONTENT
running time: 9352_µs
>>> %Run -c $EDITOR_CONTENT
running time: 9352_µs
>>> %Run -c $EDITOR_CONTENT
running time: 9350_µs
>>> %Run -c $EDITOR_CONTENT
running time: 9351_µs

```

```
MicroPython v1.15 on 2021-04-18; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
running time: 8545_µs
>>> %Run -c $EDITOR_CONTENT
running time: 8530_µs
>>> %Run -c $EDITOR_CONTENT
running time: 8529_µs
>>> 
```

CG control does not change the picture to much (at least with this code sample) V1.15 still about 9% quicker than V.16

It seems to be quite consistent over runs.

But it's not as obvious than the my original code, I still have to find out why

pidou46, 2021-07-08:
I have included GC control in my original code

```
>>> %Run -c $EDITOR_CONTENT
samples: 576
RMS: 0.2083333
running time: 31815_µs
>>> %Run -c $EDITOR_CONTENT
samples: 576
RMS: 0.0
running time: 31708_µs
>>> %Run -c $EDITOR_CONTENT
samples: 576
RMS: 0.0
running time: 31711_µs
```
```
MicroPython v1.16 on 2021-07-06; ESP32 module with ESP32

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
samples: 210
RMS: 707.6136
running time: 65441_µs
>>> %Run -c $EDITOR_CONTENT
samples: 210
RMS: 709.5591
running time: 65450_µs
>>> %Run -c $EDITOR_CONTENT
samples: 210
RMS: 709.2997
running time: 65400_µs
```
The V1.15 still run it twice as fast as V1.16

We have found out some slow down in ADC reading, but there is something else more significant.

pidou46, 2021-07-08:
Does it would be the filling of the array ?

```
from array import array
from time import ticks_us

data=array('l',200*[0])

#@micropython.viper
def fill_array():
    for i in range(200):
        data[i]=i
    

gc.collect()
gc.disable()
start_run=ticks_us()
fill_array()
end_run=ticks_us()
gc.enable()

print('running time: {}_µs'.format(end_run-start_run))
```

V1.16 : 1730_us
V1.16-viper : 895_us

V1.15 : 1730_us
V1.15-viper : 875_us

No big differences

pidou46, 2021-10-16:
Results from original post gives:
@160Mhz (default)
samples: 489
RMS: 592.5677
running time: 37100_µs

@240Mhz
samples: 567
RMS: 559.9108
running time: 32136_µs