micropython/micropython #7501: esp32.NVS missing

kdschlosser, 2021-07-07:
In the documentation it states that there is  esp32.NVS for writing to nonvolatile storage. I am attempting to use this class and I am getting this traceback.

```python
AttributeError: 'module' object has no attribute 'NVS'
```

I am using this firmware. 
esp32spiram-idf3-20210202-v1.14.bin

jimmo, 2021-07-07:
NVS was added after v1.14. Please try the latest firmware from micropython.org/download

kdschlosser, 2021-07-07:
I figured that out last night. and when I use it I get an error saying that there is no NVS available. I am using an ESP32 WROVER and I have no idea if NVS should be available on this device or not.

pumelo, 2021-07-16:
What was the exact error message?
Probably your partiton table does not have an nvs partition?
Take care ro reflash also the partition table not only the mp firmware.

kdschlosser, 2021-07-17:
It said "No support" or something along those  lines.

I have no clue how to flash partition tables.

jimmo, 2021-07-19:
> I figured that out last night. and when I use it I get an error saying that there is no NVS available. I am using an ESP32 WROVER and I have no idea if NVS should be available on this device or not.

> It said "No support" or something along those lines.

Can you show exactly the code (i.e. at the REPL) that you're using and the output and error messages?

This is working for me on 1.16 on an ESP32 board (TinyPico).

kdschlosser, 2021-07-20:
OK so since that error took place I had flashed some C++ code to the ESP32 using the Arduino IDE and now I have flashed to again with MicroPython and the NVS works properly. Would be nice to have a note in the docs letting us know that there is a limitation to the size of the key that is used to store the data.

I remember having to do this exact same  thing when I was working with an ESP8266 I would go into the REPL on it and the ANSI codes for some reason wasn't being sent properly and it wouldn't allow me to type anything I flashes that ESP8266 using the Adruino IDE and then back to MicroPython and it started to work properly. I am not sure why that is. This is not the first time I have come across this kind of behavior. Maybe something not being done when the MicroPython firmware is being flashed. Not sure.

kdschlosser, 2021-07-20:
I have another ESP32I have not used at all for anything. I will give that one a test and see what happens.

jimmo, 2021-07-20:
> I remember having to do this exact same thing when I was working with an ESP8266 I would go into the REPL on it and the ANSI codes for some reason wasn't being sent properly and it wouldn't allow me [...]

MicroPython firmware only flashes as much of flash as is required for the firmware itself, it will leave the rest of the flash untouched.

It assumes therefore that either the rest of the flash is blank (i.e. first time micropython has been installed after different firmware) or contains an existing micropython filesystem (upgrade from previous micropython version).

So in general it's always a good idea to do an erase_flash (with esptool) before installing firmware if the board isn't already running micropython.

kdschlosser, 2021-07-21:
I do that always before flashing anything to the ESP32 even if using the Arduino IDE to flash C code to it.

kdschlosser, 2021-07-21:
OK I just went and checked the NVS API for the esp32.

This is the error I got. ESP_ERR_NVS_NOT_FOUND

and this is the reason for that error Id namespace doesn’t exist yet and mode is NVS_READONLY

I have no control of setting the NVS as read only, so I am not sure why this error would occur.