micropython/micropython #7502: search() on a compiled regular expression ignores the "pos" argument

cpitclaudel, 2021-07-07:
With MicroPython v1.16 (on 2021-06-18; Raspberry Pi Pico with RP2040), I observe the followign unexpected output from .search() on a compiled regular expression:

```python
>>> import re
>>> RE = re.compile("a")
>>> print(RE.search("ab", 1) is None)
False # Should be True
```

Same result on MicroPython v1.12-1 on 2020-02-09; linux version.  Contrast with Python 3.8.10:

```python
>>> import re
>>> RE = re.compile("a")
>>> print(RE.search("ab", 1) is None)
True
```

It seems that the `pos` parameter is ignored in MicroPython?

```python
>>> print(RE.search("aa", 1).span())
(0, 1) # Should be (1, 2)
```

dpgeorge, 2021-09-04:
> It seems that the `pos` parameter is ignored in MicroPython?

That's correct, it's not implemented and is completely ignored.

I think it's actually pretty simple to implement the `pos` and `endpos` arguments.

vchahun, 2023-02-20:
This feature is actually implemented in the ure module (see extmod/modure.c#175). However, it is activated only if you set `MICROPY_PY_URE_MATCH_SPAN_START_END=1` in py/mpconfig.h Not sure why this is not on by default.
I suggest this issue should be closed.

dpgeorge, 2023-02-21:
> This feature is actually implemented in the ure module

I don't think it's been implemented.  It's the `pos` and `endpos` args in `pat.search(string, pos, endpos)`.

vchahun, 2023-02-23:
Well... please double check that. With `MICROPY_PY_URE_MATCH_SPAN_START_END` set to 1, I get the expected behavior:
```python
>>> import re
>>> RE = re.compile("a")
>>> print(RE.search("ab", 1) is None)
True
>>> print(RE.search("aa", 1).span())
(1, 2)
```
Relevant code below from modure.c (ure_exec):
```C
    #if MICROPY_PY_URE_MATCH_SPAN_START_END && !(defined(MICROPY_ENABLE_DYNRUNTIME) && MICROPY_ENABLE_DYNRUNTIME)

    if (n_args > 2) {
        const mp_obj_type_t *self_type = mp_obj_get_type(args[1]);
        mp_int_t str_len = MP_OBJ_SMALL_INT_VALUE(mp_obj_len(args[1]));
        const byte *begin = (const byte *)subj.begin;

        int pos = mp_obj_get_int(args[2]);
        if (pos >= str_len) {
            return mp_const_none;
        }
        if (pos < 0) {
            pos = 0;
        }
        const byte *pos_ptr = str_index_to_ptr(self_type, begin, len, MP_OBJ_NEW_SMALL_INT(pos), true);

        const byte *endpos_ptr = (const byte *)subj.end;
        if (n_args > 3) {
            int endpos = mp_obj_get_int(args[3]);
            if (endpos <= pos) {
                return mp_const_none;
            }
            // Will cap to length
            endpos_ptr = str_index_to_ptr(self_type, begin, len, args[3], true);
        }

        subj.begin = (const char *)pos_ptr;
        subj.end = (const char *)endpos_ptr;
    }
    #endif
```

dpgeorge, 2023-02-23:
@vchahun I cannot see that code anywhere in this repository.  Can you provide a URL to that code?

vchahun, 2023-02-26:
Wooo... I am sorry I got confused because circuitpython has this functionality and I did not check that the two modules are identical!
So here is the relevant commit in circuitpython: https://github.com/adafruit/circuitpython/commit/0865c9d38193b61898d631081161cac4d3365437
Should I submit a pull request ? I also have an implementation for re.findall...

dpgeorge, 2023-02-26:
> Should I submit a pull request ? I also have an implementation for re.findall...

In MicroPython it's not a matter of adding features because they are missing.  Of course, we could add any feature.  The hard part is deciding what features to include and what features to leave out, to keep MicroPython small yet useful.

There have not been many requests for the pos/endpos arguments (see this issue, only a few comments here).  And I don't think any requests for `re.findall()`.  So it really depends, if something is useful and can be implemented in minimal code, then we can consider adding it.