micropython/micropython #7504: Wifi Unknown Error 0x0102

kdschlosser, 2021-07-07:
In the code example below I get `Wifi Unknown Error 0x0102` when I attempt to set the  WiFi authentication mode `ap.config(authmode=authmode)`  I am not sure why that would be happening. Maybe someone can shed some light on it for me.

```python
import network

essid = 'ActiveHeadlights'
password = 'blankpassword'
authmode = 1
channel = 11
hidden = 0
max_clients = 3
ip = '192.168.0.100'
subnet = '255.255.255.0'

ap = network.WLAN(network.AP_IF)
ap.config(essid=essid)
ap.config(password=password)
ap.config(channel=channel)
ap.config(hidden=hidden)
ap.config(dhcp_hostname=essid)
ap.config(max_clients=3)
ap.config(authmode=authmode)

ap.ifconfig((ip, subnet, ip, ip))
ap.active(True)
```

kdschlosser, 2021-07-08:
again, documentation problem. There is no real documentation for how to use the access point portion of MicroPythons WiFi all of the examples really focus on being a WiFi client. One would think that activate() would be called after configuration settings are made. as it turns out that is not the case. again several hours spent digging through code and reading the ESP32 C API before I found out that activate must be called prior to setting the configuration settings..

kdschlosser, 2021-07-08:
however that was only part of the problem. I am still getting the same error when attempting to set the authmode.

kdschlosser, 2021-07-08:
I did however find out the correct meaning of the error code.

```ESP_ERR_INVALID_ARG (0x102): Invalid argument```

This is definitely a bug in the MicroPython core code for the ESP32

marcidy, 2021-07-12:
Apparently AUTH_WEP (and supposedly only AUTH_WEP) is not supported by ESP32 in (soft) ap mode.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_ap_config_t

kdschlosser, 2021-07-12:
OooOooo OK. good to know.  would be nice if it was documented and also produced an error that indicated that. maybe the removal of the WEP constant would also be a good idea.

closed 2025-03-03 by kdschlosser