micropython/micropython #7514: ESP32 ADC too noisy with RF activity

luke-jr, 2021-07-09:
The ADC readings bounce all over the place while WiFi is active. Unfortunately, setting WiFi inactive in MicroPython disassociates from the AP and kills TCP connections, so it's not a viable option when making readings very often.

Can interrupts be disabled right before ADC readings? Or maybe ESP32's "Modem Sleep" and avoid making the reading for the brief wake intervals? Any other possible solutions?

peterhinch, 2021-07-09:
You might want to raise this in [the forum](http://forum.micropython.org/index.php): there is maybe a better likelihood of reaching someone who's encountered this problem. If the RF is being picked up by the wiring to the ADC pin the answer may lie in a hardware filter. However implementing one for 2.4GHz may be difficult depending on your physical hardware,

luke-jr, 2021-07-09:
RF interferes with ADC inside the ESP32 itself. The only solution that seems to work is preventing any RF during ADC reads.

luke-jr, 2021-07-10:
For reference, Tasmota successfully reads the ADC without noise. So it's definitely possible...

luke-jr, 2021-09-08:
Noticed Tasmota uses IDF 3.3, so tried `esp32-idf3-20210202-v1.14.bin` in case it was an IDF regression, but no, this micropython also has a lot of noise. :/

dpgeorge, 2021-09-10:
> For reference, Tasmota successfully reads the ADC without noise. So it's definitely possible...

If you could find out how Tasmota does this then that could guide how it can be improved here (but be careful of licensing, Tasmota is GPL).

luke-jr, 2021-09-10:
It appears to go through some "Arduino" abstraction.

That abstraction seems to amount to:
1. First time only: adc_set_clk_div(1); adc1_config_width(3)
2. gpio_config to disable other uses (of note, including rtc_gpio_deinit) - however, on ESP32, it appears to be special-cased with a custom ~8 line implementation
4. adc1_config_channel_atten
5. adc1_get_raw

Since 4 & 5 appear to be identical to MicroPython, I am guessing it is either the adc_set_clk_div call and/or the gpio_config.

I also noticed Tasmota does 32 rapid reads and averages them, but doing the same in MicroPython doesn't appear to solve this.

luke-jr, 2021-09-10:
Is there any easy way I can test these hypothesises? Building MicroPython (especially the ESP IDF part which doesn't support my system) seems to be very complex... :/

oclyke, 2021-09-11:
May I ask what system you'd like to build on? The MicroPython team has continually made improvements to the build process, extensibility, and documentation since I started using it three+ years ago. Perhaps we could help you through the process.

The ESP IDF can be tricky but I have also noticed a significant improvement there. Perhaps if you tell us where you are having trouble. This discussion, though, would be much better to have on [the micropython forum](https://forum.micropython.org/index.php) to keep this thread focused on the ADC noise issue.

luke-jr, 2021-09-11:
POWER9 running Gentoo and zero third-party binaries (everything self-compiled, ensured by tweaking the ABI so it isn't compatible with foreign binaries).

The IDF seems to assume it can just download random stuff and run it, and explicitly doesn't support PPC64 anyway.

I do have an Xtensa cross-toolchain, if that helps...

luke-jr, 2021-09-14:
FWIW, I setup a x86 system to play around with this on and got it building MicroPython fine. But none of the things mentioned above helped :(