micropython/micropython #7516: When using the mpremote tool, must specify the port

yangwc, 2021-07-09:
When I using the mpremote tool, must specify the port number in every instruction at every time. Otherwise, it can't work. 
Only if the port number is specified, it works OK, the mpremote version is 0.0.5, python version is 3.6 
[image]

peterhinch, 2021-07-09:
Perhaps you could provide more details: what OS you're running and what target you are connecting to. I regularly connect to RP2, ESP32 and Pyboards under Linux with:
```bash
$ mpremote mount .
```

dpgeorge, 2022-03-07:
`mpremote` has been updated to 0.2.0.  Please try again with the latest version, it should be working.

closed 2022-03-07 by dpgeorge