micropython/micropython #7523: @micropython.native wrong result for bytes(iterable)

twisteroidambassador, 2021-07-11:
@micropython.native
    def native_unpack(packed: int):
        return bytes((255, 255 if packed & 1 else 0, 255 if packed & 2 else 0, 255 if packed & 4 else 0))
    
    
    def unpack(packed: int):
        return bytes((255, 255 if packed & 1 else 0, 255 if packed & 2 else 0, 255 if packed & 4 else 0))

The above two functions return different result when run on Pyboard 1.1 with firmware v1.16. The return value from the undecorated function is correct, while the native decorated function returns erroneous results:

    >>> native_test.unpack(0)
    b'\xff\x00\x00\x00'
    >>> native_test.native_unpack(0)
    b'\xff\xff\x00\x00'

jimmo, 2021-07-12:
Here's a cut-down example showing what's happening

```python
@micropython.native
def native_unpack(packed: int):
    return 1, 2, 3, 255 if packed & 1 else 0
```

with generated code (for x86-64, but the issue is in emitnative, not the arch-specific).

```
       0:   55                      push   %rbp
       1:   53                      push   %rbx
       2:   41 54                   push   %r12
       4:   41 55                   push   %r13
       6:   48 83 ec 58             sub    $0x58,%rsp
       a:   4c 8b 6f 18             mov    0x18(%rdi),%r13
       e:   49 8b 6d 08             mov    0x8(%r13),%rbp
      12:   48 89 3c 24             mov    %rdi,(%rsp)
      16:   bf bf 00 00 00          mov    $0xbf,%edi
      1b:   48 89 7c 24 08          mov    %rdi,0x8(%rsp)
      20:   bf 06 00 00 00          mov    $0x6,%edi
      25:   48 89 7c 24 18          mov    %rdi,0x18(%rsp)
      2a:   48 89 e7                mov    %rsp,%rdi
      2d:   48 8b 85 68 01 00 00    mov    0x168(%rbp),%rax
      34:   ff d0                   call   *%rax
      36:   48 8b 5c 24 50          mov    0x50(%rsp),%rbx
      3b:   ba 03 00 00 00          mov    $0x3,%edx
      40:   48 89 de                mov    %rbx,%rsi
      43:   bf 18 00 00 00          mov    $0x18,%edi
      48:   48 8b 85 90 00 00 00    mov    0x90(%rbp),%rax
      4f:   ff d0                   call   *%rax                  # <-- mp_binary_op
      51:   48 89 c7                mov    %rax,%rdi
      54:   48 8b 85 80 00 00 00    mov    0x80(%rbp),%rax
      5b:   ff d0                   call   *%rax                  # <-- mp_obj_is_true (result in eax=0)
      5d:   b8 03 00 00 00          mov    $0x3,%eax              # _
      62:   48 89 44 24 28          mov    %rax,0x28(%rsp)        # |
      67:   b8 05 00 00 00          mov    $0x5,%eax              # |
      6c:   48 89 44 24 30          mov    %rax,0x30(%rsp)        # | < store the immediate into the stack (3, 5, 7 are the mp_obj_t small ints for 1, 2, 3)
      71:   b8 07 00 00 00          mov    $0x7,%eax              # |
      76:   48 89 44 24 38          mov    %rax,0x38(%rsp)        # -
      7b:   84 c0                   test   %al,%al                # <-- eax is clobbered (now non-zero)
      7d:   0f 84 0f 00 00 00       je     0x92
      83:   b8 ff 01 00 00          mov    $0x1ff,%eax
      88:   48 89 44 24 40          mov    %rax,0x40(%rsp)
      8d:   e9 0a 00 00 00          jmp    0x9c
      92:   b8 01 00 00 00          mov    $0x1,%eax
      97:   48 89 44 24 40          mov    %rax,0x40(%rsp)
      9c:   48 8d 74 24 28          lea    0x28(%rsp),%rsi
      a1:   bf 04 00 00 00          mov    $0x4,%edi
      a6:   48 8b 85 98 00 00 00    mov    0x98(%rbp),%rax
      ad:   ff d0                   call   *%rax
      af:   e9 00 00 00 00          jmp    0xb4
      b4:   48 83 ec a8             sub    $0xffffffffffffffa8,%rsp
      b8:   41 5d                   pop    %r13
      ba:   41 5c                   pop    %r12
      bc:   5b                      pop    %rbx
      bd:   5d                      pop    %rbp
      be:   c3                      ret    

```

So `emit_native_jump_helper` is calling `need_stack_settled` between `calling mp_obj_is_true` and emitting the jump. Because the immediates being used to generate the tuple haven't been written to the stack yet, `need_stack_settled` writes them out before the jump, but it uses `REG_TEMP0` (rax) to do so, which is clobbering the return value of `mp_obj_is_true`.

@dpgeorge I _think_ the solution here is to use `REG_TEMP1` in need_stack_settled instead?

@twisteroidambassador as a temporary workaround, putting each value into a variable then returning `(a,b,c,d)` will avoid the issue.

dpgeorge, 2021-07-12:
Thanks @twisteroidambassador for the report, and @jimmo for the diagnosis.

> I _think_ the solution here is to use `REG_TEMP1` in need_stack_settled instead?

Maybe... but not sure.  Will need to look through it in detail.

tannewt, 2021-07-16:
Jim and I talked over this issue and fix on my Deep Dive stream today: https://www.youtube.com/watch?v=bciRImf0_Ts Thanks to Jim for taking the time!

commit 0e3752e8 (2021-07-19): py/emitnative: Ensure stack settling is safe mid-branch.

And add a test for the case where REG_RET could be in use.

Fixes #7523.
closed 2021-07-19 by dpgeorge