micropython/micropython #7524: Bug with defining I2C object with both I2C Bus ID and pins on ESP32

CurlyTaleGames, 2021-07-11:
I have the issue pretty well documented here with some screenshots:
[https://forum.micropython.org/viewtopic.php?f=15&t=10635&p=59456](https://forum.micropython.org/viewtopic.php?f=15&t=10635&p=59456)

When I defined an I2C object and passed in the I2C Bus ID and also defined the SCL and SDA pins I couldn't get I2C frequencies faster than 15khz

When I removed the Bus ID I was getting speeds of 85khz

When I used the Bus ID and removed the SCL and SDA pins and switched to the default pins I got 100khz

```python

# This one would only work with a frequency of 15khz max:
i2c = I2C(0, scl=Pin(22), sda=Pin(21), freq=15000)

# This one would work but I was only getting 85khz when I defined 100khz
i2c = I2C(scl=Pin(22), sda=Pin(21), freq=100000)

# This one is working correctly and I'm seeing 100khz
i2c = I2C(0, freq=100000)
# I2C bus 0 default pins:
# scl = 18
# sda = 19

```

jonnor, 2024-10-01:
Does this still happen on a recent MicroPython, such as 1.23 ?

jonnor, 2025-03-03:
Missing information to reproduce.

closed 2025-03-03 by jonnor