micropython/micropython #7525: 1.13+ doesn't compile on FreeBSD 13.0

yggdr, 2021-07-11:
Starting with micropython 1.13 I cannot compile it on FreeBSD 13 anymore. 1.12 worked fine. Installation methods were installation via pyenv's `python-build`, as well as the `ports/unix` method described in the README.

```
[...]
CC ../../py/formatfloat.c
In file included from ../../py/formatfloat.c:33:
../../py/formatfloat.c: In function 'mp_format_float':
../../py/formatfloat.c:131:20: error: conversion from 'double' to 'float' may change value [-Werror=float-conversion]
  131 |     if (fp_signbit(f) && !fp_isnan(f)) {
      |                    ^
../../py/formatfloat.c:131:9: note: in expansion of macro 'fp_signbit'
  131 |     if (fp_signbit(f) && !fp_isnan(f)) {
      |         ^~~~~~~~~~
../../py/formatfloat.c:131:36: error: conversion from 'double' to 'float' may change value [-Werror=float-conversion]
  131 |     if (fp_signbit(f) && !fp_isnan(f)) {
      |                                    ^
../../py/formatfloat.c:131:27: note: in expansion of macro 'fp_isnan'
  131 |     if (fp_signbit(f) && !fp_isnan(f)) {
      |                           ^~~~~~~~
../../py/formatfloat.c:146:22: error: conversion from 'double' to 'float' may change value [-Werror=float-conversion]
  146 |         if (fp_isinf(f)) {
      |                      ^
../../py/formatfloat.c:146:13: note: in expansion of macro 'fp_isinf'
  146 |         if (fp_isinf(f)) {
      |             ^~~~~~~~
../../py/formatfloat.c:151:29: error: conversion from 'double' to 'float' may change value [-Werror=float-conversion]
  151 |         } else if (fp_isnan(f)) {
      |                             ^
../../py/formatfloat.c:151:20: note: in expansion of macro 'fp_isnan'
  151 |         } else if (fp_isnan(f)) {
      |                    ^~~~~~~~
cc1: all warnings being treated as errors
gmake: *** [../../py/mkrules.mk:77: build-standard/py/formatfloat.o] Error 1
```

dpgeorge, 2021-07-12:
Thanks for the report, I can confirm the issue using a FreeBSD virtual box (FreeBSD 12.0 on 64-bit x86-64).

As far as I can tell, this is a bug in gcc...??  The following simple program shows the same issue:
```c
#include <stdio.h>
#include <math.h>

int main() {
    double x = 13.4;
    printf("%f %d\n", fpclassify(x));
    return 0;
}
```
compiled with `gcc -Wfloat-conversion test.c`.  gcc version is 8.3.0.

dpgeorge, 2021-07-12:
gcc has had similar bugs in the past, see eg https://gcc.gnu.org/bugzilla/show_bug.cgi?id=61501

yggdr, 2021-07-12:
Wow, a gcc bug was not what I expected…

fyi:
```
$ gcc --version
gcc (FreeBSD Ports Collection) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ cc --version
FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Target: x86_64-unknown-freebsd13.0
Thread model: posix
InstalledDir: /usr/bin
```

As an experiment I checked out tag v1.12, `make clean`ed everything, then changed `py/mkenv.mk` to set make's CC variable to cc instead of gcc, to test if clang can compile micropython. Then I `make V=1` in `mpy-cross` first, followed by `ports/unix`. This works and produces a runable micropython.

Then I redid the same thing in current master (925878b2f83e09987bd1688746b92898ee8bbcac), now I get a bit further, until this happens:
```
[…]
cc -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I../../lib/mp-readline -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../extmod/axtls-include -DMICROPY_PY_B
TREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I/usr/local/include  -DMICROPY_PY_FFI=1 -Os -DNDEBUG -fdata-sections -ffunction-sections -Ivariants/standard  -g -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -DMICROPY_MODULE_FROZEN_S
TR -c -MD -o build-standard/extmod/uos_dupterm.o ../../extmod/uos_dupterm.c
CC ../../lib/embed/abort_.c
cc -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I../../lib/mp-readline -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../extmod/axtls-include -DMICROPY_PY_B
TREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I/usr/local/include  -DMICROPY_PY_FFI=1 -Os -DNDEBUG -fdata-sections -ffunction-sections -Ivariants/standard  -g -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -DMICROPY_MODULE_FROZEN_S
TR -c -MD -o build-standard/lib/embed/abort_.o ../../lib/embed/abort_.c
CC ../../lib/utils/printf.c
cc -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I../../lib/mp-readline -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../extmod/axtls-include -DMICROPY_PY_B
TREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I/usr/local/include  -DMICROPY_PY_FFI=1 -Os -DNDEBUG -fdata-sections -ffunction-sections -Ivariants/standard  -g -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -DMICROPY_MODULE_FROZEN_S
TR -c -MD -o build-standard/lib/utils/printf.o ../../lib/utils/printf.c
python3 ../../tools/makemanifest.py -o build-standard/frozen_content.c -v "MPY_DIR=../.." -v "MPY_LIB_DIR=../../../micropython-lib" -v "PORT_DIR=/storage/code/micropython/ports/unix" -v "BOARD_DIR=" -b "build-standard" -f"-mcache-lookup-bc" --mpy-tool-flags="" variants/manifest.py
MPY upip.py
error compiling /storage/code/micropython/tools/upip.py:
make: *** [../../py/mkrules.mk:142: build-standard/frozen_content.c] Error 1
```

Would the 13MB `mpy-cross.core` this creates be of help here?

dpgeorge, 2021-07-12:
> Would the 13MB `mpy-cross.core` this creates be of help here?

Not really, would be better to reproduce the error fully in a virtual machine or similar.

FWIW I could build master (mpy-cross and unix) on FreeBSD 12.0 with `gmake CFLAGS_EXTRA=-Wno-float_conversion`

yggdr, 2021-07-12:
Yay, rebuilding everything with that option worked :) (it's `no-float-conversion`, not `no-float_conversion`)

dpgeorge, 2021-07-12:
I don't think there's much more we can do here, but I'm happy to leave the issue open for a bit to see if anything comes up.

closed 2024-03-15 by dpgeorge