micropython/micropython #7528: esp32 call f_open() fun error

Spring641, 2021-07-12:
I tried to open a file error, how to solve，is f_open()

FIL* boot_output_file;
FIL file_pointer;
boot_output_file = &file_pointer;

// Get the base filesystem.
FATFS *fs = &((fs_user_mount_t *) MP_STATE_VM(vfs_mount_table)->obj)->fatfs;

if (f_open(fs, boot_output_file, "boot.py", FA_READ) == FR_OK)
{
	f_close(boot_output_file);
	printf("f_open ok\r\n");
	
}else{
	printf("f_open error\r\n");
}

Spring641, 2021-07-12:
ELF file SHA256: 8fba20119d615e89

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002577e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x11e4
load:0x4004c000,len:0x8f0
load:0x40050000,len:0x2938
entry 0x4004c1b0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400bb43a  PS      : 0x00060030  A0      : 0x800a40f9  A1      : 0x3ffd4220  
A2      : 0x00000001  A3      : 0x400bb43a  A4      : 0x3ffd4250  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x0000002f  A8      : 0x3f023e8c  A9      : 0x00000006  
A10     : 0xfffffffe  A11     : 0x00000000  A12     : 0x3ffd4100  A13     : 0x3fd8c050  
A14     : 0x00000006  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x3ffd4100  LEND    : 0x3fd8c050  LCOUNT  : 0x40026c4d  

Backtrace:0x400bb437:0x3ffd42200x400a40f6:0x3ffd4240 0x400a5410:0x3ffd4290 0x40089f9d:0x3ffd42f0

jimmo, 2021-07-12:
I'd suggest adding some more `printf` to narrow it down further (check that `fs` is not NULL, etc).

On ESP32 by default the main filesystem will be LittleFS not FAT, so I suspect that might be the issue?

Also, rather than using the fatfs API directly, I'd suggest using the VFS layer. (e.g. start with `mp_vfs_lookup_path`).

Spring641, 2021-07-13:
> I'd suggest adding some more `printf` to narrow it down further (check that `fs` is not NULL, etc).
>
> On ESP32 by default the main filesystem will be LittleFS not FAT, so I suspect that might be the [...]

mp_vfs_open ok

closed 2021-07-13 by jimmo