micropython/micropython #7533: How to compile ESP32 firmware with IDF4.0 or above

hyzgit, 2021-07-12:
In 4.0, the command idf.py has been used to compile, not make

https://github.com/micropython/micropython/tree/master/ports/esp32

Building the firmware
The MicroPython cross-compiler must be built to pre-compile some of the built-in scripts to bytecode. This can be done by (from the root of this repository):

$ make -C mpy-cross
Then to build MicroPython for the ESP32 run:

$ cd ports/esp32
$ make submodules
$ make

jimmo, 2021-07-12:
@hyzgit   Could you please clarify what you're raising an issue for? Is there a documentation issue? Thanks

hyzgit, 2021-07-12:
@jimmo Thank you, this is my first attempt to compile MicroPython firmware for ESP32,However, when I tried to use the instructions in the following page, I found that there was a problem at compile time. The make command was not available,After looking for some relevant content on the Internet,IDF4.3 is not compiled with the make command. 
https://github.com/micropython/micropython/tree/master/ports/esp32/README.md

Don't know how to go on now，Can someone update this document so that it can match IDF4.3

jimmo, 2021-07-12:
> I found that there was a problem at compile time.
What's the error you're seeing?

> Can someone update this document so that it can match IDF4.3
Have you tried using 4.2? In the README it says "Currently MicroPython supports v4.0.2, v4.1.1 and v4.2".

> The make command was not available
Even though the ESP32 port now uses CMake, the make command still works (the Makefile just invokes CMake for you). You do need to have make (and CMake) installed on your system (as well as having set up the IDF with install.sh and export.sh as described in README)

FYI this might be better suited for the forum -- https://forum.micropython.org/

hyzgit, 2021-07-12:
@jimmo Thank you, I do not have make and cmake installed, I will install these two first and then try, thank you

dpgeorge, 2021-07-12:
> Don't know how to go on now，Can someone update this document so that it can match IDF4.3

Compiling is the same for all IDF 4.x versions.  You can use the convenience Makefile and `make`, or `idf.py` on its own (but look at the `Makefile` for hints on what parameters to pass `idf.py`).

hyzgit, 2021-07-12:
@jimmo 
cmake.exe mpy-cross
CMake Error: The source directory "D:/GitCodes/esp32/micropython/mpy-cross" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

dpgeorge, 2021-07-12:
You need to build `mpy-cross` using normal `make`.

hyzgit, 2021-07-13:
@dpgeorge Mpy-cross doesn't have to compile, does it ?

The following is the error I received when compiling ESP32 firmware

D:\GitCodes\esp32\micropython\ports\esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory d:\gitcodes\esp32\micropython\ports\esp32\build
Executing "ninja all"...
[138/1253] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure
cmd.exe /C "cd /D D:\GitCodes\esp32\micropython\ports\esp32\build\bootloader && C:\Users\hyj\.espressif\tools\cmake\3.16.4\bin\cmake.exe -DSDKCONFIG=D:/GitCodes/esp32/micropython/ports/esp32/build/sdkconfig -DIDF_PATH=D:/GitCodes/esp32/esp-idf -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\hyj\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe -DEXTRA_COMPONENT_DIRS=D:/GitCodes/esp32/esp-idf/components/bootloader -DLEGACY_INCLUDE_COMMON_HEADERS= -GNinja D:/GitCodes/esp32/esp-idf/components/bootloader/subproject && C:\Users\hyj\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E touch D:/GitCodes/esp32/micropython/ports/esp32/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
CMake Error: The source "D:/GitCodes/esp32/esp-idf/components/bootloader/subproject/CMakeLists.txt" does not match the source "D:/Other/esp32/esp-idf/components/bootloader/subproject/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
[155/1253] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/rsn_supp/wpa.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

D:\GitCodes\esp32\micropython\ports\esp32>

dpgeorge, 2021-07-13:
> Mpy-cross doesn't have to compile, does it

Yes it needs to be built.

You need to follow the instructions in `ports/esp32/README.md`.

> CMake Error: The source "D:/GitCodes/esp32/esp-idf/components/bootloader/subproject/CMakeLists.txt" does not match the source "D:/Other/esp32/esp-idf/components/bootloader/subproject/CMakeLists.txt" used to generate cache. Re-run cmake with a different source directory.

To fix that error, you probably just need to start from scratch (or remove the build directory completely).

hyzgit, 2021-07-14:
@dpgeorge This is what I did in ports/esp32/README.md , But some steps cannot be followed in the documentation，

Building the firmware
The MicroPython cross-compiler must be built to pre-compile some of the built-in scripts to bytecode. This can be done by (from the root of this repository):

All the previous steps are fine. It's always down here

$ make -C mpy-cross
Then to build MicroPython for the ESP32 run:

$ cd ports/esp32
$ make submodules
$ make

You will find that the make command cannot be executed，Do you want to change all “make” to “idf.py”，Or can this document be updated? If idf.py is used, how should it be changed? Can we also update the documentation to work with idf.py ， Because the novice, according to the makefile to modify their own idf.py command, can not know how to change，Thank you very much.

dpgeorge, 2021-07-14:
> You will find that the make command cannot be executed

To help diagnose the problem, can you please show exactly what commands you execute and exactly what the output is?

hyzgit, 2021-07-14:
@dpgeorge 

D:\GitCodes\esp32\esp-idf>export.bat
Setting IDF_PATH: D:\GitCodes\esp32\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\hyj\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin
    C:\Users\hyj\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\hyj\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\hyj\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\hyj\.espressif\tools\cmake\3.16.4\bin
    C:\Users\hyj\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin
    C:\Users\hyj\.espressif\tools\ninja\1.10.0\
    C:\Users\hyj\.espressif\tools\idf-exe\1.0.1\
    C:\Users\hyj\.espressif\tools\ccache\3.7\
    C:\Users\hyj\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\hyj\.espressif\python_env\idf4.2_py3.8_env\Scripts
    D:\GitCodes\esp32\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from D:\GitCodes\esp32\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

D:\GitCodes\esp32\esp-idf>cd ..\micropython

D:\GitCodes\esp32\micropython>make -C mpy-cross
MAKE Version 5.43  Copyright (c) 1987, 2019 Embarcadero Technologies, Inc.
Incorrect command line argument: -C

Syntax: MAKE [options ...] target[s]
    -B                Builds all targets regardless of dependency dates
    -Dsymbol[=string] Defines symbol [equal to string]
    -Idirectory       Names an include directory
    -K                Keeps (does not erase) temporary files created by MAKE
    -N                Increases MAKE's compatibility with NMAKE
    -Wfilename        Writes MAKE to filename updating all non-string options
    -Usymbol          Undefine symbol
    -ffilename        Uses filename as the MAKEFILE
    -a                Performs auto-dependency checks for include files
    -c                Caches auto-dependency information
    -e                Ignores redefinition of environment variable macros
    -i                Ignores errors returned by commands
    -l+               Enables use of long command lines
    -m                Displays the date and time stamp of each file
    -n                Prints commands but does not do them
    -p                Displays all macro definitions and implicit rules
    -q                Returns zero if target is up-to-date and nonzero
                      if it is not (for use in batch files)
    -r                Ignores rules and macros defined in BUILTINS.MAK
    -s                Silent, does not print commands before doing them
    -? or -h          Prints this message
      Options marked with '+' are on by default. To turn off a default
      option follow it by a '-', for example: -a-
D:\GitCodes\esp32\micropython>

marcidy, 2021-07-17:
Try running `make` with no arguments from inside the mpy-cross directory

hyzgit, 2021-07-18:
@marcidy 
D:\GitCodes\esp32\micropython>make -C mpy-cross
'make' is not recognized as an internal or external command,
operable program or batch file.

D:\GitCodes\esp32\micropython>cd mpy-cross

D:\GitCodes\esp32\micropython\mpy-cross>make
'make' is not recognized as an internal or external command,
operable program or batch file.

D:\GitCodes\esp32\micropython\mpy-cross>

hyzgit, 2021-07-18:
D:\GitCodes\esp32\esp-idf>export.bat
Setting IDF_PATH: D:\GitCodes\esp32\esp-idf

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Users\hyj\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\hyj\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\hyj\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\hyj\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\hyj\.espressif\tools\cmake\3.16.4\bin;C:\Users\hyj\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\hyj\.espressif\tools\ninja\1.10.0\;C:\Users\hyj\.espressif\tools\idf-exe\1.0.1\;C:\Users\hyj\.espressif\tools\ccache\3.7\;C:\Users\hyj\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\hyj\.espressif\python_env\idf4.2_py3.8_env\Scripts;D:\GitCodes\esp32\esp-idf\tools;C:\Program Files (x86)\Eziriz\.NET Reactor;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\nodejs\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Java\jdk1.8.0_271\bin;C:\Program Files\Java\jdk1.8.0_271\jre\bin;C:\Program Files\Calibre2\;C:\Program Files\Git\cmd;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Users\hyj\AppData\Local\Yarn\bin;;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\dotnet\;C:\Users\hyj\AppData\Local\Microsoft\WindowsApps;C:\Users\hyj\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\hyj\AppData\Roaming\npm;C:\Users\hyj\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\Scripts;C:\Users\hyj\.dotnet\tools

Checking if Python packages are up to date...
Python requirements from D:\GitCodes\esp32\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

D:\GitCodes\esp32\esp-idf>cd ..\micropython

D:\GitCodes\esp32\micropython>make
'make' is not recognized as an internal or external command,
operable program or batch file.

D:\GitCodes\esp32\micropython>

marcidy, 2021-07-18:
what happened to `make`?  Your previous comment had it installed.

```
D:\GitCodes\esp32\micropython>make -C mpy-cross
MAKE Version 5.43 Copyright (c) 1987, 2019 Embarcadero Technologies, Inc.
Incorrect command line argument: -C
```
Here it's a recognized program.  What happened?

If your version of `make` doesn't support `-C`, you don't need it, you can run `make` from inside the mpy-cross directory.  But if you lost `make` then you need to fix that first.

hyzgit, 2021-07-18:
@marcidy This is another computer of mine that has Embarcadero installed, so the make command, by default, uses this software,
I changed to a new computer, followed the  ports/esp32/README.md steps again, downloaded and installed, the result is no make command, prompt like above.

marcidy, 2021-07-18:
that's something you need to solve, clearly `make` is a requirement.

hyzgit, 2021-07-18:
> that's something you need to solve, clearly `make` is a requirement.

Yes, but I also follow the steps to operate. Why does it prompt that there is no make? Is it the new version of esp-idf that uses idf.py to compile instead of make?After export. BAT, it is prompted to compile with idf.py build

Here's the full tip

D:\GitCodes\esp32\esp-idf>export.bat
Setting IDF_PATH: D:\GitCodes\esp32\esp-idf

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Users\hyj.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\hyj.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\hyj.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\hyj.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\hyj.espressif\tools\cmake\3.16.4\bin;C:\Users\hyj.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\hyj.espressif\tools\ninja\1.10.0;C:\Users\hyj.espressif\tools\idf-exe\1.0.1;C:\Users\hyj.espressif\tools\ccache\3.7;C:\Users\hyj.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\hyj.espressif\python_env\idf4.2_py3.8_env\Scripts;D:\GitCodes\esp32\esp-idf\tools;C:\Program Files (x86)\Eziriz.NET Reactor;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\Microsoft SQL Server\130\Tools\Binn;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn;C:\Program Files\nodejs;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\Java\jdk1.8.0_271\bin;C:\Program Files\Java\jdk1.8.0_271\jre\bin;C:\Program Files\Calibre2;C:\Program Files\Git\cmd;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Users\hyj\AppData\Local\Yarn\bin;;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\dotnet;C:\Users\hyj\AppData\Local\Microsoft\WindowsApps;C:\Users\hyj\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\hyj\AppData\Roaming\npm;C:\Users\hyj\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\Scripts;C:\Users\hyj.dotnet\tools

Checking if Python packages are up to date...
Python requirements from D:\GitCodes\esp32\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

idf.py build

D:\GitCodes\esp32\esp-idf>cd ..\micropython

hyzgit, 2021-07-18:
D:\GitCodes\esp32\micropython>idf.py --version
ESP-IDF v4.2

D:\GitCodes\esp32\micropython>

marcidy, 2021-07-18:
I have no idea, that's something only you can figure out.  Try typing `idf.py build` in `ports/esp32` to see if it starts a build or fails on the check for `make`.  I suspect it will.

marcidy, 2021-07-18:
also note the requirements on the README state that you need `make` installed, so either way you need to install it.

hyzgit, 2021-07-18:
> also note the requirements on the README state that you need `make` installed, so either way you need to install it.

Make installation is not included in the README steps, so it needs to be installed separately ?

marcidy, 2021-07-18:
Yes.  Do you want to compile the firmware, or just use the firmware?  If you just want to use it, download from here (specifically `esp32-20210623-v1.16.bin`), and follow the instructions to install esptool.py to flash it:  https://github.com/espressif/esptool .  This will be much easier to get started.  

If you really want to develop / compile the firmware yourself, then you need the requirements.  It will probably be much easier to use Windows Subsystem for Linux instead of trying to do it under Windows directly.

hyzgit, 2021-07-18:
> Yes. Do you want to compile the firmware, or just use the firmware? If you just want to use it, download from https://micropython.org/download/esp32/ (specifically `esp32-20210623-v1.16.bin`), and follow the instructions to install esptool.py to flash it: https://github.com/espressif/esptool . This will be much easier to get started.
> 
> If you really want to develop / compile the firmware yourself, then you need the requirements. It will probably be much easier to use Windows Subsystem for Linux instead of trying to do it under Windows directly.

There is no problem with the use of the code, now I want to compile by myself, add some of my own code in the firmware, increase the security of the code (prevent copying products)
Thank you, as well as the other friends above, for switching to Ubuntu and it has been successfully compiled.

closed 2021-07-19 by jimmo