micropython/micropython #7545: SAMD User modules are not included in firmware

v923z, 2021-07-16:
I would like to move an issue (https://github.com/v923z/micropython-ulab/issues/424) upstream, because I have the feeling that it has to do with the core. It seems that user modules are not included in the firmware on the SAMD port. Here is an example:

```bash
v923z:samd$ make BOARD=ADAFRUIT_ITSYBITSY_M4_EXPRESS
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr
GEN build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr/mpversion.h
...
LINK build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 102120	     24	   1340	 103484	  1943c	build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
Converting to uf2, output size: 204288, start address: 0x4000
Wrote 204288 bytes to build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.uf2.
```
And the same with `ulab`

```bash
v923z:samd$ make BOARD=ADAFRUIT_ITSYBITSY_M4_EXPRESS USER_C_MODULES=../../../ulab all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../ulab/code
mkdir -p build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr
...
LINK build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 104032	     24	   1340	 105396	  19bb4	build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
Converting to uf2, output size: 208384, start address: 0x4000
Wrote 208384 bytes to build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.uf2.
```
Curiously, while the initial step of the build process indicates that code will be included from `ulab`, there is absolutely no trace of it later, while, at the same time, the firmware infinitesimally increased in size. I would have expected to see the compilation output

```bash
CC ../../../ulab/code/scipy/linalg/linalg.c
CC ../../../ulab/code/scipy/optimize/optimize.c
CC ../../../ulab/code/scipy/signal/signal.c
CC ../../../ulab/code/scipy/special/special.c
CC ../../../ulab/code/ndarray_operators.c
CC ../../../ulab/code/ulab_tools.c
CC ../../../ulab/code/ndarray.c
CC ../../../ulab/code/numpy/ndarray/ndarray_iter.c
CC ../../../ulab/code/ndarray_properties.c
CC ../../../ulab/code/numpy/approx/approx.c
CC ../../../ulab/code/numpy/compare/compare.c
CC ../../../ulab/code/ulab_create.c
CC ../../../ulab/code/numpy/fft/fft.c
CC ../../../ulab/code/numpy/fft/fft_tools.c
CC ../../../ulab/code/numpy/filter/filter.c
CC ../../../ulab/code/numpy/linalg/linalg.c
CC ../../../ulab/code/numpy/linalg/linalg_tools.c
CC ../../../ulab/code/numpy/numerical/numerical.c
CC ../../../ulab/code/numpy/poly/poly.c
CC ../../../ulab/code/numpy/stats/stats.c
CC ../../../ulab/code/numpy/transform/transform.c
CC ../../../ulab/code/numpy/vector/vector.c
CC ../../../ulab/code/numpy/numpy.c
CC ../../../ulab/code/scipy/scipy.c
CC ../../../ulab/code/user/user.c
CC ../../../ulab/code/utils/utils.c
CC ../../../ulab/code/ulab.c
```

I have checked this with the `unix`, and `stm32` ports, there everything works fine. Thinking that perhaps `ulab` simply doesn't fit the flash of the board, I also tried the trivial factorial example from the docs, and that is not included either.

I am not against patching this up on my side, but having established that this probelm is most probably not `ulab`-related, I think the issue belongs here.

dpgeorge, 2021-07-22:
Thanks for the report, indeed it's a problem with the samd port not supporting user C modules.

Should be fixed by #7557.  Would you be able to test that PR?

v923z, 2021-07-22:
> Should be fixed by #7557. 

Many thanks!

> Would you be able to test that PR?

I don't have a piece of hardware, I can only look at the firmware size, and infer something from that.

dpgeorge, 2021-07-22:
> I don't have a piece of hardware, I can only look at the firmware size, and infer something from that.

Ok, no problem.

dpgeorge, 2021-07-22:
Fixed by a5ac3d5645e99a897f3213149587a43b62779c35 (and tested on a MINISAM_M4).

closed 2021-07-22 by dpgeorge