micropython/micropython #7546: unknown warning option '-Wimplicit-fallthrough=0'

oclyke, 2021-07-17:
Just went to build the ```dev``` variant of the unix port and got this error:

```
CC ../../lib/btstack/src/ad_parser.c
error: unknown warning option '-Wimplicit-fallthrough=0'; did you mean '-Wimplicit-fallthrough'? [-Werror,-Wunknown-warning-option]
make: *** [build-incident/lib/btstack/src/ad_parser.o] Error 1
```

Building on: macOS 10.15.7 (19H1217)
From commit: 8be29b9

I modified the file ```extmod/btstack/btstack.mk``` to omit the offending option entirely and was able to proceed through a successful build. No warnings were generated.

I understand this flag was added recently (#6578). Could this be a linux vs. osx issue?

dpgeorge, 2021-07-17:
This looks like a difference between gcc and clang.  Probably that warning can be omitted when building with clang (eg put in the `ifneq ($(CC),clang)` part of the Makefile).

Might be a good idea to build `micropython-dev` on the mac CI.

dpgeorge, 2021-07-17:
See #7547 for a fix, and also enabling of `micropython-dev` build on macOS CI.

dpgeorge, 2021-07-17:
Should be fixed by 74db526cf059d1bc30be38c28c36cf5d1156ae53

closed 2021-07-17 by dpgeorge