micropython/micropython #7550: DHCP server config

kdschlosser, 2021-07-19:
I have written a very basic DNS server for MicroPython that I am wanting to run on the ESP32. I also wrote a DHCP server to run as well. Upon starting my DHCP server I would get an error for the port being in use. After digging through code for a while I discovered that MicroPython as a default starts a DHCP server then the ESP32 is set to be an access point. 

The problem is that there is no way to adjust any of the DHCP server settings that is built into the ESP programming. I am specifically looking to add a DNS server to the DHCP offer packets. I would have thought that this would be set using the DNS server IP addresses that are supplied when using WLAN.ifconfig. and they are not. 

It would be exceedingly helpful if the ESP32 WiFi is set to AP_IF and WLAN.ifconfig is called that the DHCP offer packets include the option for DNS servers and those servers are set to the servers that are passed in WLAN.ifconfig. May as well set the gateway in the same manner. If those IP address are blank `""` then the DNS and gateway options should be removed from the DHCP server.

Even better would be to add complete control of the internal DHCP including the ability to turn it on and off. Not everyone that uses the ESP32 has a need for the DHCP server anyhow and it should be an optional thing to turn on.

I would add it myself but I am not proficient in C/C++ and I am not familiar with the MicroPython API to be able to do so.

pumelo, 2021-07-22:
I raised adding the dns in dhcp on the esp32 here #6014.
The PR should still be functioning and should allow to set the dns server sent in the dhcp messages.

zcattacz, 2024-05-02:
We have a isolated wireless network for the workshop workers to use phones to submit form. Once in a dnsmasq debugging session I noticed the gateway is bombarded with all sorts of failed connecrion tries from different phones and apps. If they don't have the dhcp set gateway, such noise can be neutralized without any extra processing power in an MCU context.

I'd like to have that option in a micropython case. and let the user navigate manually with their bookmark.

kdschlosser, 2024-05-02:
I should add support for DHCP. I could write a user c module that would be used instead of the WiFi built into MicroPython. In the years since I opened this issue I have spent a great deal of time working with the underlying C code and writing modules to expose other portions of the ESPIDF. Like the ability to create buffer objects that use DMA memory and exposing things like TWAI and the esp_lcd component to the python side of things.

closed 2025-03-03 by kdschlosser