micropython/micropython #7560: mimxrt: Question about extending the test base

robert-hh, 2021-07-22:
In out attempt to extend the test base we procured a MIMXRT1050 board. That caused problems, since the old board is not available any more, and the new board is by deafult configured for a hyperflash chip instead a QSPI flash. The layout is different, especially the page size is 512 byte instead of 256 byte. For LFS that means that progsize has to be set differently, and Python code must have a means to determine the page size. A iotcl call could used for that, but it has to be defined: Currently the is no code provided for that. Do you know a code foreseen for this function, or should we just grap one, like 7?

robert-hh, 2021-07-22:
As another option, I added a QSTR to that specific block device called PAGE_SIZE, which will be set board-specific.

dpgeorge, 2021-07-22:
> especially the page size is 512 byte instead of 256 byte

Do you mean the minimum write size is now 512 bytes?  But the erase size is still 4096?  Can you remind me why `progsize` needs to match this value (or be larger than it, I guess)?

> As another option, I added a QSTR to that specific block device called PAGE_SIZE, which will be set board-specific.

Yeah, that's not a bad way to do it... but a new ioctl might be more general.

robert-hh, 2021-07-22:
For that flash, the page size if 512 byte, the sector size is 256 kByte, and the full flash size is 64MB. As far as I understand it, writing to the flash has to be done in page size items. I tried keeping progsize=256, but that failed. So it seems that progsize has match the page size, but I can do further testing.
I changed the QSTR to a bound methed. That way I can refer to a value in the flash config block instead of a #define. So it's defined at a single place. A IOCTL call would do the same.

robert-hh, 2021-07-22:
The restriction may be caused by the simple writeblocks() implementation in this port. It assumes that both the offset and data size is a multiple of the flash page size. From what I noticed earlier, vfs requests writes for lfs with a multiple of progsize for offset and length (length was actually always progsize). For FAT the offset was a multiple of 512. So I could change the writeblocks() implementation allowing writes of partial pages. But that would double the size of managing data in lfs to about 32k. That may be worse than having larger progsize items.

robert-hh, 2021-07-23:
I changed the writeblocks() code to allow partial page writes. Therefore the is no change required for vfs or the block driver interface,

closed 2021-07-23 by robert-hh