micropython/micropython #7565: Combine ESP32C3 firmware bug

Walkline80, 2021-07-23:
ESP32C3 bootloader offset address changed to `0x0`, but the param `OFFSET_BOOTLOADER` in file `/ports/esp32/makeimg.py` still is `0x1000`

There're two ways to flash the firmware

1. Using `esptool.py`
```bash
/home/walkline/.espressif/python_env/idf4.3_py3.6_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS3 -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 build-GENERIC_C3/bootloader/bootloader.bin 0x8000 build-GENERIC_C3/partition_table/partition-table.bin 0x10000 build-GENERIC_C3/micropython.bin 
```
2. Using combining tools, for example `flash_download_tools` to combine 3 bins with corrected offset address

```
0x0 bootloader.bin
0x8000 partition-table.bin
0x10000 micropython.bin
```

dpgeorge, 2021-08-09:
Thanks for the report.  See #7632 for a fix.

dpgeorge, 2021-08-10:
Fixed by 3835f5f597a002149d85423dcc72371e8d8bbcc5

closed 2021-08-10 by dpgeorge