micropython/micropython #7585: REPL and `input` skip non-ascii characters of the input

aivarannamaa, 2021-07-25:
A simple beginners' exercise of asking user's name (`name = input("Enter your name: ")`) can cause confusion when the name contains non-ascii chars, which are not echoed and don't make it to the interpreter.

The same goes to entering anything to the REPL. In order to rule out problems with serial terminal, I experimented with Pyserial against RPi Pico (MP 1.16):

```python
import serial

s = serial.Serial("/dev/ttyACM0")
s.write("name = 'mõdu';print('Name:', name, len(name))\r\n".encode("utf-8"))
while True:
    print(s.readline().decode("utf-8"), end="")
```

The result was:
```
name = 'mdu';print('Name:', name, len(name))
Name: mdu 3
```

As you see, the letter "õ" didn't make it through.

Same problem occurs with ESP-32 and Unix ports. On the other hand -- CircuitPython 7.0.0a3 handles non-ascii input nicely.

The problem does not occur when reading input with `sys.stdin.readline`.

dpgeorge, 2021-07-27:
Thanks for the report.  It would be good to improve utf-8 support (and maybe see what CircuitPython does for starters).

This is closely related to #2789.

dpgeorge, 2021-07-27:
See improvement in CircuitPython here: https://github.com/adafruit/circuitpython/pull/3243