micropython/micropython #7586: release 1.16 submodules problem

szampardi, 2021-07-26:
hi folks,

FYI I get this when running `git clone --branch=v1.16 --depth=1 --recurse-submodules --shallow-submodules "https://github.com/micropython/micropython.git" "/opt/micropython"`. Works fine with `master`.

```
#11 294.3 Submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out 'b7b26f50d00072812aec8453f643e24bafedccb5'
#11 298.2 From https://github.com/FreeRTOS/FreeRTOS
#11 298.2  * branch            c40a6da2e4cb8042b56d1b174051cbbe9813781a -> FETCH_HEAD
#11 299.8 Submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/FreeRTOS': checked out 'c40a6da2e4cb8042b56d1b174051cbbe9813781a'
#11 299.8 Submodule 'FreeRTOS/Source' (https://github.com/FreeRTOS/FreeRTOS-Kernel.git) registered for path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/FreeRTOS/FreeRTOS/Source'
#11 299.8 Cloning into '/opt/micropython/lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/FreeRTOS/FreeRTOS/Source'...
#11 302.9 From https://github.com/FreeRTOS/FreeRTOS-Kernel
#11 302.9  * branch            10bbbcf0b9bcab1869b3bac3987d32116f351718 -> FETCH_HEAD
#11 303.0 Submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/FreeRTOS/FreeRTOS/Source': checked out '10bbbcf0b9bcab1869b3bac3987d32116f351718'
#11 304.7 error: Server does not allow request for unadvertised object 0192fe773ec28e11f66ec76f4e827fbb58b7e257
#11 304.7 Fetched in submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/lwip', but it did not contain 0192fe773ec28e11f66ec76f4e827fbb58b7e257. Direct fetching of that commit failed.
#11 304.7 Failed to recurse into submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb'
```

dpgeorge, 2021-07-26:
You don't need a recursive clone, only top-level submodules are needed.

szampardi, 2021-07-26:
oh! thanks!

dpgeorge, 2021-07-26:
> Works fine with `master`.

In that case there's not much that can be done here, v1.16 is already released.  And in the next release a recursive clone, although not needed, should work.

closed 2021-07-26 by dpgeorge

szampardi, 2021-07-26:
@dpgeorge mind elaborating on why fixing 1.16 is not possible? 

> not much that can be done here, v1.16 is already released

It doesn't look like any of the developers of this project is signing their commits, so you're just one `git tag -f` away from having a proper 1.16 release tag. Cheers.

dpgeorge, 2021-07-27:
> mind elaborating on why fixing 1.16 is not possible?

We can make a 1.16.1 release if really necessary.

szampardi, 2021-07-27:
it is not necessary, not for me at least, just a proper-ness concern. Thank you anyway.

I wouldn't want to have broken modules or anything of the kind tagged in my latest "stable" release, for the projects I work on.

dpgeorge, 2021-07-27:
> I wouldn't want to have broken modules or anything of the kind tagged in my latest "stable" release, for the projects I work on.

Yeah, fair enough (but there are always bugs, that's unavoidable really).  None of the CI here uses a recursive clone so this issue was never seen, and never a problem.  I'm not even sure what the fix is.