micropython/micropython #7592: How to switch submodule 'axtls' branch?

water5, 2021-07-26:
From `/micropython/`, `git submodule update --init --recursive`, return:

> fatal: remote error: upload-pack: not our ref 531cab9c278c947d268bd4c94ecab9153a961b43
> Fetched in submodule path 'axtls', but it did not contain 531cab9c278c947d268bd4c94ecab9153a961b43. Direct fetching of that commit failed.

From `/micropython/lib/axtls`, `git ls-remote`, return:

> From https://github.com/pfalcon/axtls
> 4315d76a6761ad332795a7a13b2c0449c4b00aa8	HEAD
> 4315d76a6761ad332795a7a13b2c0449c4b00aa8	refs/heads/master
> 3c6b62b718427c03c345ee15be9bb66959a34ccd	refs/heads/micropython
> f38bdcb22c23d2721adf0f6b659e57be21f13e93	refs/heads/micropython-2.1.3
> 131056b2399ac22fe9f36d1d98f9c49016ffeb60	refs/heads/ssl_eagain
> 4af1d046ad263745f2ba2fa4a362291a4aac9f97	refs/pull/3/head
> 971eaf449b7dcb99e263a098c2bf87afa9be14ef	refs/pull/3/merge
> c421c13c63e5d659273ccff38e8000365cc87206	refs/pull/4/head
> cbee4aa80fb3f363897fe576dd747189c365194e	refs/pull/4/merge
> 131056b2399ac22fe9f36d1d98f9c49016ffeb60	refs/pull/5/head
> 3dc2cad63fb99788f48c73dab4b6893dd12038b2	refs/pull/5/merge
> 3348622ec6c6803263d676c32d61a4f11082b7f4	refs/pull/6/head
> a122fe9fc9937feebb35e99669476a46128510bf	refs/pull/6/merge
> c2690dc100f272cb01375dcf2f2be7105d0fb7c0	refs/tags/v1.8.2

From `/micropython/lib/axtls`, `git branch`, return:

> * (HEAD detached at 43a6e6b)
>   master

How to switch to `531cab9c278c947d268bd4c94ecab9153a961b43` ?

mcauser, 2021-07-26:
In .gitmodules it recently changed from
https://github.com/pfalcon/axtls
to
https://github.com/micropython/axtls

https://github.com/micropython/micropython/commit/2e634d5ead9bfce8cdacf317470534a16d8e6631

dpgeorge, 2021-07-26:
Try this:
```
$ git submodule sync lib/axtls
$ git submodule update lib/axtls
```

closed 2021-07-27 by water5

dlech, 2021-09-01:
Would it help if we pushed commit 531cab9c278c947d268bd4c94ecab9153a961b43 from the old submodule as a branch to the new submodule to avoid this error in the first place?