micropython/micropython #7597: run main module on unix startup

oclyke, 2021-07-30:
asking because [last time](https://github.com/micropython/micropython/issues/7440) i tried to do something like this i learned a much better way from the community and i hope there is a similar solution in this case...

i would like to make a variant of the unix port that runs ```main.py``` on startup to mimic the behavior of the microcontroller ports. i managed to do this with **some kludgy feeling steps:**
* modify ```ports/unix/main.c``` to include these lines:
	``` c
	#include "lib/utils/pyexec.h"
	...
    // run boot-up scripts
    pyexec_frozen_module("_boot.py");
    pyexec_file_if_exists("boot.py");
    if (pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {
        pyexec_file_if_exists("main.py");
    }
	```
	which just feels bad from a maintenance perspective - i'd prefer not to diverge from the main repo except in variant specific areas
* add ```LIB_SRC_C_EXTRA += utils/pyexec.c``` to ```mpconfigvariant.h``` (this feels like by far the least concerning modification required)
* add ```MICROPY_HW_BOARD_NAME``` and ```MICROPY_HW_MCU_NAME``` definitions to ```mpconfigvariant.h``` - this just doesn't feel right for the unix port, but was necessary to resolve compilation errors in ```lib/utils.pyexec.c```

**so, my question(s):**
* is there a simpler way to do what i am trying to do?
* is it intentional that compiling ```lib/utils/pyexec.c``` requires preprocessor defines that seem very microcontroller specific?
* how much effort would one expect it to be to make a PR that fixes the above situation?

dlech, 2021-07-30:
Why not just run `micropython main.py` from the command line?

Or add `#!/usr/bin/env micropython` to the top of the `main.py` file, make the file executable (`chmod +x main.py`) and run `./main.py`?

oclyke, 2021-08-20:
^^^ that's exactly what i was looking for, i just didn't know at the time! thanks a ton @dlech 😃

for future reference the [unix port quick ref documentation](https://docs.micropython.org/en/latest/unix/quickref.html) clearly describes the [```<script>```](https://docs.micropython.org/en/latest/unix/quickref.html#cmdoption-arg-script) argument as well as several other useful bits

coming from an embedded perspective i didn't consider the executable could accept ***arguments*** how nifty is that??

closed 2021-08-20 by oclyke