micropython/micropython #7605: ESP8266 socket.recv(1024) doesn't get all data

PawelZieba01, 2021-08-01:
Hello,
I have a problem with receiving http requests on my ESP8266 module (MicroPython v1.16).
I've started listening server and  sending to them request that contains 598 chatacters. Then I get only 584 bytes on my ESP console :(

Have you ever seen this or similr problem? I've checked it on ESP32 module (MicroPython v1.16) and it works good :|
Thanjs for help and your time :)

Here is my simple code:
```
import network

import socket

wlan = network.WLAN(network.STA_IF)

wlan.active(True)
wlan.ifconfig(("192.168.1.150", "255.255.255.255", "192.168.1.1", "8.8.8.8"))
wlan.connect("my-ssid", "my-pass")

while wlan.isconnected() == False:
    pass
print("ok")

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('', 80))
s.listen(3)

while True:
    conn, addr = s.accept()
    request = conn.recv(1024)
    conn.sendall('HTTP/1.1 200 OK\nConnection: close\nServer: nanoWiPy\nContent-Type: text/html\n\n')
    
    request = str(request)
    print(request)
```

And this is request I'm sending:
```
POST /x HTTP/1.1\r\nHost: 192.168.1.150\r\nConnection: keep-alive\r\nContent-Length: 141\r\nUser-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.107 Safari/537.36\r\nDNT: 1\r\nContent-Type: text/plain\r\nAccept: */*\r\nOrigin: chrome-extension://gmmkjpcadciiokjpikmkkmapphbmdjok\r\nAccept-Encoding: gzip, deflate\r\nAccept-Language: pl-PL,pl;q=0.9,en-US;q=0.8,en;q=0.7\r\n\r\ndddddddddddddddddddxxxxxxxxxxxxxcccccccccccccccccccccccccccccccxxxxxxxxxxaaaaaaaaaaxxxxxxxxxxzzzzzzzzzzzzzaaaaaaaaaaavvvvv12345677dsdsdsdsd77
```

And I'm getting this:
```
b'POST /x HTTP/1.1\r\nHost: 192.168.1.150\r\nConnection: keep-alive\r\nContent-Length: 141\r\nUser-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.107 Safari/537.36\r\nDNT: 1\r\nContent-Type: text/plain\r\nAccept: */*\r\nOrigin: chrome-extension://gmmkjpcadciiokjpikmkkmapphbmdjok\r\nAccept-Encoding: gzip, deflate\r\nAccept-Language: pl-PL,pl;q=0.9,en-US;q=0.8,en;q=0.7\r\n\r\ndddddddddddddddddddxxxxxxxxxxxxxcccccccccccccccccccccccccccccccxxxxxxxxxxaaaaaaaaaaxxxxxxxxxxzzzzzzzzzzzzzaaaaaaaaaaavvvvv12345'
```

dpgeorge, 2021-08-01:
When you use `socket.recv(n)` it will return up to `n` bytes.  But it may return less, and it may not get all the available data.  You must keep calling `recv()` until it closes or you know you have all the data.

PawelZieba01, 2021-08-01:
Thank you for answer :)
I will try to modify my program according to your suggestion.

PawelZieba01, 2021-08-01:
Ok, I've put second line with `conn.recv(1024)` and now I'm getting all my data. 
```
...
while True:
    conn, addr = s.accept()
    request1 = conn.recv(1024)
    request2 = conn.recv(1024)
    
    conn.sendall('HTTP/1.1 200 OK\nConnection: close\nServer: nanoWiPy\nContent-Type: text/html\n\n')
    conn.close()
    
    request1 = request1.decode("utf-8")
    request2 = request2.decode("utf-8")
    print(request1)
    print(request2)
...
```

But why on ESP32 i can use single `conn.recv(1024)`?
Does it result from firmware or hardware difference?

peterhinch, 2021-08-02:
The amount of data you receive in a single call depends on the other end of the link. How quickly it responds to the request, communications conditions between ends of the link, and suchlike. You need a loop (with a timeout) because the time to receive all your data can be arbitrarily long.

Incidentally, support queries like this are best posted in [the forum](http://forum.micropython.org/index.php).

closed 2021-08-03 by dpgeorge