micropython/micropython #7607: ESP32 RTC_VALID_EXT_PINS not valid for ESP32 S2/S3

stephanelsmith, 2021-08-02:
modesp32.h defines the valid RTC_VALID_EXT_PINS suitable for Pico D4.  S2 has a different pinout.  

For example, pin 10 is not valid on Pico D4 (Flash memory), but is a valid RTC pin on S2.  If I'm not mistaken, I believe this should work on S2, but getting 'invalid pin' error. 

```
o = Pin(10, Pin.IN, pull=Pin.PULL_UP)
esp32.wake_on_ext0(pin = o, level = esp32.WAKEUP_ALL_LOW) #single pin
```

Thank you!

dpgeorge, 2021-08-05:
If you have a suggestion how to fix this, please tell.

stephanelsmith, 2021-08-06:
@dpgeorge I think the change is a philosophical in nature.  I'm looking at a narrow slice of micropython, at the moment (S2 and soon S3) implementations, and so this may not be the right big picture approach.  

Right now in modesp32.h, the code is enforcing the correct pins for external interrupts, the RTC pins are the valid ones.  In my opinion, the code shouldn't need to have this level of enforcement.  The RTC_IS_VALID_EXT_PIN macro is inherently a moving target and will always be an issue on every new variant.  I think it should be removed.

Relevant piece:
`#define RTC_VALID_EXT_PINS \
    ( \
    (1ll << 0) | \
    (1ll << 2) | \
    (1ll << 4) | \
    (1ll << 12) | \
    (1ll << 13) | \
    (1ll << 14) | \
    (1ll << 15) | \
    (1ll << 25) | \
    (1ll << 26) | \
    (1ll << 27) | \
    (1ll << 32) | \
    (1ll << 33) | \
    (1ll << 34) | \
    (1ll << 35) | \
    (1ll << 36) | \
    (1ll << 37) | \
    (1ll << 38) | \
    (1ll << 39)   \
    )
#define RTC_LAST_EXT_PIN 39
//#define RTC_IS_VALID_EXT_PIN(pin_id) ((1ll << (pin_id)) & RTC_VALID_EXT_PINS)
#define RTC_IS_VALID_EXT_PIN(pin_id) 1
`

Maybe the docs (https://docs.micropython.org/en/latest/library/esp32.html) should state what valid pins are for esp32.wake_on_ext0 and esp32.wake_on_ext1 (must be RTC pins).  

My change is simple, I just disabled the RTC_IS_VALID_EXT_PIN.  This macro seems exclusively used for policing wake source pin selection.
`
//#define RTC_IS_VALID_EXT_PIN(pin_id) ((1ll << (pin_id)) & RTC_VALID_EXT_PINS)
#define RTC_IS_VALID_EXT_PIN(pin_id) 1
`

I ran into a similar issue regarding S2/S3 SPI implementation, where SPI2 and SPI3 are used, which is different from the current ESP32 port.  The issue was already raise and solved here:
https://github.com/micropython/micropython/pull/7291
The change works for me.  I would encourage bringing that into the mainline for future S2/S3 users!

Again, Micropython is a huge and awesome project, and no sure if my work around is a fit for the project at large.

dpgeorge, 2021-12-15:
Ok, I understand the problem.  IMO it's useful to prevent the user from trying to use a pin when it won't work for the purpose (RTC wakeup), otherwise users can get quite lost.  But of course it shouldn't prevent using a pin when that pin is valid!

If the IDF already catches illegal pins used for wakeup then we can remove `RTC_IS_VALID_EXT_PIN()` without losing the ability to warn the user (the IDF will warn, print out a statement).

Otherwise we could define different `RTC_VALID_EXT_PINS` for each variant, S2, S3, C3 etc.

dpgeorge, 2022-04-14:
Fixed by 988827b85a9b09f1164e101ea2520562dd31634e

closed 2022-04-14 by dpgeorge