micropython/micropython #7612: esp32: Allow configuring the build to use out of tree boards

mocleiri, 2021-08-04:
The esp32 cmake build supports setting a **MICROPY_BOARD_DIR** variable to the absolute path of an out of tree board directory.

This allows you to have the board configurations outside of the micropython submodule which makes tracking the latest micropython easier.

The gnu makefile bridge doesn't provide a way to pass through a custom **BOARD_DIR** at the moment to the idf.py command line.

dpgeorge, 2021-08-04:
I'm not sure exactly what the issue/question/suggestion here is, but you can see an example of an out-of-tree ESP32 board here: https://github.com/micropython/micropython-example-boards/tree/main/boards/CUSTOM_ESP32

mocleiri, 2021-08-04:
Sorry I was supposed to file a pull request for this at the same time.  

Thanks for linking to the examples for how to make out of tree boards.  That's exactly what I needed.

My approach was to add a new option into the gnu makefile which would allow setting the MICROPY_BOARD_DIR.

https://github.com/mocleiri/micropython/commit/6232fea5700690bc8130f08fd0adae6e2b0bd736

That way you could run the build like this (from ports/esp32):
make BOARD_DIR=$BASE_DIR/boards/esp32/MICROLITE BOARD=MICROLITE USER_C_MODULES=$BASE_DIR/micropython-modules/micropython.cmake

Yesterday I saw the MICROPY_BOARD_DIR option but couldn't locate the examples you linked to do I made my own way.

But I like the way the examples do it better than my way so I'm going to migrate my board configurations and close this issue instead of making a pull request.

closed 2021-08-04 by mocleiri