micropython/micropython #7621: unix port: building requires python?

winterismute, 2021-08-05:
I wanted to take a look at the project because it seems really interesting for some use-cases, and decided to clone it and to build the "unix" port on my osx machine, just as a test. I was also interested in understanding what is the minimum set of files needed to build that.
At some point, I saw "make" calling:
```
/usr/local/bin/python3: can't open file '/Users/rbonfiglioli/mine/micropython_mine/ports/unix/../../tools/makemanifest.py': [Errno 2] No such file or directory
```
I was a bit surprised: does building the micropython interpreter require to have some python runtime already installed?

dlech, 2021-08-05:
Yes, it is used to generate some files during the build process.

stinos, 2021-08-05:
Note this is mentioned at the end of the first paragraph in the README

winterismute, 2021-08-05:
> Note this is mentioned at the end of the first paragraph in the README
Thanks, I noticed it now.

Do you have suggestions on how to proceed if compiling on a system with no python available? I assume you do not maintain this flow, however you might have an idea how complex it might be to move the work done by the python scripts to eg. bash scripts... Or also, would it be possible to pre-generate those files in an environment different than the one where want to build?

dlech, 2021-08-05:
> Do you have suggestions on how to proceed if compiling on a system with no python available?

Install Python? If you can compile MicroPython on the system, then you can probably also compile Python on the system.

jimmo, 2021-08-06:
> Do you have suggestions on how to proceed if compiling on a system with no python available? I assume you do not maintain this flow, however you might have an idea how complex it might be to move the work done by the python scripts to eg. bash scripts...

@winterismute The work done by these Python scripts is relatively complex and would not be feasible to implement in Bash. They could be re-written in C, but Python is a lot more convenient.

One of the main things to look at is the QSTR generation, which is all done in Python. https://docs.micropython.org/en/latest/develop/qstr.html

> Or also, would it be possible to pre-generate those files in an environment different than the one where want to build?

Yes, this would be possible, with some work and assuming the environments were identical. It might also be possible with some work, as @dlech suggested, to use MicroPython instead of CPython to run these scripts.

winterismute, 2021-08-06:
Hi @jimmo, thank you a lot for your reply.
The thing is: I need to approach a quite constrained environment, both in terms of HW resources eg memory, but also with a rudimentary OS, which does not expose a lot of the POSIX/standard C lib. Since I was looking at options to run python there, I thought micropython could be a good fit, however you and/or @dlech seem to hint at the fact that it might be actually harder to build on an arbitrary platform than CPython? Up to now I did not do a full assessment yet and I was just checking out micropython for the sake of it...

jimmo, 2021-08-06:
> The thing is: I need to approach a quite constrained environment, 

It sounds like you want to cross-compile for your target rather than building on-device (i.e. more like how we compile for the bare-metal & RTOS ports). i.e. rather than trying to pre-generate the generated files on Linux/Mac and then build on-device, build the binary entirely on Linux/Mac.

stinos, 2021-08-06:
> seem to hint at the fact that it might be actually harder to build on an arbitrary platform than CPython?

Having tried this once, for a platform which wasn't not supported yet by both, I think the MicroPython codebase itself should be easier to get built than CPython in general because it is smaller and normally depens only standard C99 and standard C library. However yes the preprocesing step needs some extra attention. Out of interest: have you tried CPython already on your platform?

dpgeorge, 2021-09-04:
It would be nice if MicroPython didn't need CPython to build.  One way to do this would be to get a bootstrapped version of MicroPython running on the target system (something very minimal), then use it to build the main/final MicroPython.

To that end, it would be a good goal to make the main Python generation scripts in this repo run under MicroPython (and maybe early on that was the case).

[Rewriting the generation scripts in C would be another way to eliminate CPython, as @jimmo says, but I don't think anyone here has the time to do that.]

jonnor, 2024-10-11:
The primary method to get MicroPython onto a platform which does not have CPython, would be to cross-compile.
As mentioned, increasing MicroPython compatibility of the build scripts, to support bootstrapping via MicroPython instead of CPython is welcomed. But it does not seem like anyone has worked actively on this the last years, so I propose to close this issue.

dpgeorge, 2024-10-15:
I think it's a worthy goal, to have a way to build the unix port with minimal system dependencies.  I actually made a little progress in this direction but it's not working yet.

Let's keep this issue open to keep this idea alive.