micropython/micropython #7622: Question About Licensing

hello6800, 2021-08-05:
Hiya, I was viewing the license for this project and saw that some licensed were (L)GPL. I heard that if _any_ file is (L)GPL, it must be open source, so does my project need to be open source?

jimmo, 2021-08-06:
Disclaimer: I'm the author of that section of the LICENSE file, but I am not a license expert and you should consult your own legal advice if this is important to your project. Here's some background to explain the wording in the LICENSE file.

No (L)GPL code is included in the final binary for any MicroPython port. The LICENSE states: "Note that (L)GPL-licensed code listed below is only used during the build process and is not part of the compiled source code."

The justification here regarding build tools is that it's like using GCC to compile the firmware -- GCC is licensed under the GPL, but programs compiled with GCC are not.

I am not aware of any issue with having GPL licensed code in the same repository, the issue is whether it's used in the final "program". A piece of code has a license, and in the case of GPL (and MIT!) that license then applies to the final program generated from that code. The repo itself has a top-level license which functions as a summary/guideline (and to indicate the "default" license for code in that repo). In MicroPython we explicitly put a license at the top of every file.

To be more specific, there are two relevant bits of code:
 - axtls/config -- this is a small configuration tool that is used as part of the axtls build. Note that axtls itself (which we do link against on some ports) is BSD-style. http://axtls.sourceforge.net/README/index.html
 - tools/dfu.py -- this is a standalone tool, used to convert compiled firmware into .dfu files (i.e. it's like a compiler).

hello6800, 2021-08-06:
I see, but let’s say I gave the source code to he public, wi

jimmo, 2021-08-06:
> I see, but let’s say I gave the source code to he public, without an open source license, I”m guessing this doesn’t need to be (L)GPL licensed since the code hasn’t been built?

I'm not quite sure I follow sorry. Which source code? All of the MicroPython source code is MIT licensed, so this would be against the terms of the MIT license too (and this would require stripping the license from each file individually too).

hello6800, 2021-08-06:
Code that I make with MicroPython, sorry for not specifying.

jimmo, 2021-08-06:
> Code that I make with MicroPython, sorry for not specifying.

Can you provide an example? What is "code" in this case?

hello6800, 2021-08-06:
Take for example this code:
```python
import machine
thing = machine.Pin(14, machine.Pin.OUT)
```

dlech, 2021-08-06:
> All of the MicroPython source code is MIT license

The original source for [micropython/ports/esp8266/espneopixel.c](https://github.com/micropython/micropython/blob/4d9e657f0ee881f4a41093ab89ec91d03613744d/ports/esp8266/espneopixel.c#L1) is GPL3, so doesn't that make that file GPL3 *and* MIT?

dlech, 2021-08-06:
> Take for example this code

The code you write yourself can be any license you like. 

My interpretation is that (L)GPL copyleft restrictions only apply to code that is linked at compile time. Since MicroPython is an interpreted language, there is no linking involved so these restriction don't apply.

hello6800, 2021-08-06:
Ah ok, thank you!

closed 2021-08-06 by hello6800

iabdalkader, 2021-08-20:
> > No (L)GPL code is included in the final binary for any MicroPython port.
> 
> The original source for [micropython/ports/esp8266/espneopixel.c](https://github.com/micropython/micropython/blob/4d9e657f0ee881f4a41093ab89ec91d03613744d/ports/esp8266/espneopixel.c#L1) is LGPL-3, so doesn't that make that file LGPL-3 _and_ MIT?

I've been thinking about this recently, would it make sense to add a top level define to disable (L)GPL code? Something like MICROPY_NO_GPL (1). This will allow (L)GPL code to exist in the repo, and not get compiled into the final binary by default, but can be enabled by ports/boards if needed.

jimmo, 2021-08-20:
> I've been thinking about this recently, would it make sense to add a top level define to disable (L)GPL code? Something like MICROPY_NO_GPL (1). This will allow (L)GPL code to exist in the repo, and not get compiled into the final binary by default, but can be enabled by ports/boards if needed.

I can see the benefit to this, and for example ffmpeg does something along these lines to disable including GPL code (in an otherwise LGPL project). I do think GPL in an otherwise MIT project isn't quite the same thing from a "spirit of the law perspective" but I don't really know.

For MicroPython there's a big difference between doing this in the core, extmod, ports, and boards.

My preference would be to handle incompatible-licensed code explicitly in a different repo. So for example a board definition (or port) that wants to use GPL code could live out-of-tree.

> The original source for micropython/ports/esp8266/espneopixel.c is LGPL-3, so doesn't that make that file LGPL-3 and MIT?

FWIW this (and the esp32 version) have now been removed. @dlech

iabdalkader, 2021-08-20:
> For MicroPython there's a big difference between doing this in the core, extmod, ports, and boards.

I was thinking more for drivers/libraries etc...

> My preference would be to handle incompatible-licensed code explicitly in a different repo

It could be an LGPL submodule, anyway since there's none of that right now it's not needed at this point.