micropython/micropython #7633: Enhance capabilities of mp_arg_parse_all

jepler, 2021-08-09:
In CircuitPython, we lean a lot more heavily on `mp_arg_parse_all` than micropython typically does. We'd like to do even more with it, such as validating that arguments fall into a given range, applying a custom transformation to an argument, and maybe more; we strongly suspect that we can save overall firmware size by adding and then using these facilities.

Here are the extra validation types I considered (with some cross-pollination from @dhalbert). We'll probably prototype this in CircuitPython and analyze which ones actually save us flash space, before trying to provide a version that is potentially able to be pulled into MicroPython:

```c
typedef enum {
    // The corresponding argument is converted and stored in .u_bool
    MP_ARG_BOOL      = 0x001,

    // The corresponding object is stored in u_obj
    MP_ARG_OBJ       = 0x002,

    // The corresponding argument is converted to an integer (or a float, if
    // MP_ARG_FLOAT is or'd in) and stored in u_int or u_float.  The default
    // is the mp_arg_val_t's u_int or u_float.
    MP_ARG_NUMBER    = 0x003,
    MP_ARG_INT       = MP_ARG_NUMBER,

    // The corresponding object is passed to a function, which may return
    // a populated mp_arg_val_t or raise an exception
    MP_ARG_FUNC      = 0x004,

    // The corresponding value must be an instance of the type stored in u_obj, or
    // one of its subtypes
    MP_ARG_TYPE      = 0x005,

    // The corresponding value must be an instance of exactly the type stored in u_obj
    MP_ARG_XTYPE     = 0x006,

    // The corresponding value must be an integer or float in the signed
    // inclusive range given by u_slo..u_shi.  The default is None if
    // MP_ARG_OR_NONE is specified, else -1 if MP_ARG_OR_MINUS1 is specified,
    // else u_slo.
    MP_ARG_RANGE16   = 0x007,

    // The corresponding value must be an integer or float in the unsigned
    // inclusive range given by u_ulo..u_uhi.  The default is None if
    // MP_ARG_OR_NONE is specified, else -1 if MP_ARG_OR_MINUS1 is specified,
    // else u_lo.

    MP_ARG_URANGE16  = 0x008,
    // The corresponding value must be an integer or float in the signed
    // inclusive 0..u_int or u..u_float.  The default is none if MP_ARG_OR_NONE
    // is specified, else -1 if MP_ARG_OR_MINUS1 is specified, else 0.
    MP_ARG_0_TO_N    = 0x009,

    // The corresponding value must be an integer or float in the signed
    // inclusive 1..u_int or u..u_float.  The default is none if MP_ARG_OR_NONE
    // is specified, else -1 if MP_ARG_OR_MINUS1 is specified, else 1.
    MP_ARG_1_TO_N    = 0x00a,

    // The corresponding value must be an a power of two in the range
    // 2^u_ulo .. 2^u_uhi inclusive. The default is None if MP_ARG_OR_NONE is
    // specified, else -1 if MP_ARG_OR_MINUS1 is specified, else 2^u_ulo.
    // Incompatible with MP_ARG_FLOAT.
    MP_ARG_POW2      = 0x00b,

    // A bitmask that includes all enumerated values above
    MP_ARG_KIND_MASK = 0x01f,
    // A numeric argument (MP_ARG_NUMBER, MP_ARG_RANGExx, etc) is a float,
    // not an int
    MP_ARG_FLOAT     = 0x020,

    // None is also accepted.  If a u_int is to be stored, it is stored as -1.
    // If a u_float is to be stored, it is stored as a NaN.  None will be used
    // as the default value if the defval field's meaning is overloaded.
    MP_ARG_OR_NONE   = 0x040,

    // -1 is also accepted.  -1 will be used as the default value if the defval
    // field's meaning is overloaded, unless MP_ARG_OR_NONE is also specified.
    MP_ARG_OR_MINUS1 = 0x080,

    MP_ARG_REQUIRED  = 0x100,
    MP_ARG_KW_ONLY   = 0x200,

    // Instead of a single argument, a sequence where all items conform to the
    // value specification is required.  MP_ARG_FUNC becomes a predicate only,
    // the transformed value is discarded.
    MP_ARG_SEQUENCE_OF = 0x400,
} mp_arg_flag_t;
```

You could validate that a pin object was passed:
```c
{ MP_QSTR_scl, MP_ARG_REQUIRED | MP_ARG_XTYPE, { .u_obj = &mcu_pin_type}},
```
or even that a sequence of pins was passed:
```c
{ MP_QSTR_data_pins, MP_ARG_REQUIRED | MP_ARG_XTYPE | MP_ARG_SEQUENCE_OF, { .u_obj = &mcu_pin_type}},
```

Or that a buffer request is a power of 2 from (1<<3) to (1<<8) inclusive:
```c
{ MP_QSTR_buffer_size, MP_ARG_REQUIRED | MP_ARG_POW2, { .u_ulo = 3, .u_uhi = 8 }
```

dlech, 2021-08-09:
If we could do this and save firmware space at the same time, that would be amazing. We use this quite a bit in [Pybricks](https://github/pybricks/pybricks-micropython) as well.

Some special cases we have are:
- allow float for int arguments
- if a numeric argument is out of range, clamp it to an in range value rather than raise ValueError
  - example 1: if the range is 0 to 100, then negative numbers become 0 and > 100 becomes 100
  - example 2: for degrees we sometime want the value modulo 360
- We have an enum-like type where it would be nice to get the numeric value instead of the enum object
 
---

> ```c
> MP_ARG_FUNC
> ```

Does this mean `callable` is true? (If so, this it is not necessary a function).

jepler, 2021-08-09:
> ```c
> MP_ARG_FUNC
> ```
> Does this mean callable is true? (If so, this it is not necessary a function).

No, rather it means that the corresponding C function stored in `defval.u_func` is called to validate the argument and return an `mp_obj_arg_val_t`.  For instance, you mentioned the use-case of accepting an int, or a float rounded/cast to an int.  Except for the issue of a default value, you could implement this by having a function `arg_number_as_int` with an implementation something like
```c
mp_obj_arg_val_t arg_number_as_int(const struct _mp_arg_t *arginfo, const mp_obj_t obj) {
    mp_obj_arg_val_t result = {.u_int = mp_obj_get_int(obj); };
    return result;
}
```
and elsewhere
```c
{ MP_QSTR_number, MP_ARG_REQUIRED | MP_ARG_FUNC, { .u_func = arg_number_as_int } },
```

dlech, 2021-08-09:
Cool, it sounds like this one would handle pretty much all of our use cases.

jepler, 2021-08-28:
I took a second stab at this, first implementing MP_ARG_FLOAT and then MP_ARG_FUNC, MP_ARG_TYPE and MP_ARG_ARRAY_OF. This is at https://github.com/adafruit/circuitpython/compare/main...jepler:arg-parse-2?expand=1

I found that

    There weren't as many places as I thought to validate ranges
    On non-express m0 boards, there weren't enough uses of MP_ARG_FLOAT to justify the code bloat
    Even on express m0 boards (specifically circuitplayground_express) there weren't enough uses of MP_ARG_FUNC & MP_ARG_TYPE to justify the code bloat.

This is disappointing, I expected there to be small but real wins on the low end. Looks like that is not the case.

Perhaps I made some errors in implementation, but those are my findings.

closed 2021-08-28 by jepler