micropython/micropython #7638: Conversion from const char* to mp_rom_error_text_t ?

iabdalkader, 2021-08-11:
The compiler can't always evaluate the strcmp tree generated by `MP_ERROR_TEXT(x)` at compile time if `x` is determined at runtime (for example a call to an old style xxx_strerror function):

```C
mp_raise_msg(&mp_type_OSError, MP_ERROR_TEXT(ffs_strerror(res)));
```

This results in a copy of the tree for every invocation of MP_ERROR_TEXT, so the return result can't be compressed and must be cast to `mp_rom_error_text_t` instead. This should be safe for now, but will it change in the future ? And if so, shouldn't there be a function or macro to hide this conversion since all exception formally accept `mp_rom_error_text_t` arg ? Something like:

```c
#define MP_ERROR_TEXT_FROM_STR(x) (mp_rom_error_text_t) x
```

dpgeorge, 2021-08-11:
> shouldn't there be a function or macro to hide this conversion since all exception formally accept `mp_rom_error_text_t` arg ? Something like

Yes I would agree, there should be a macro for this case.

You can see in `extmod/modussl_mbedtls.c:mbedtls_raise_error` that it does something like this and explicitly constructs the str arg for the exception.  That case should also arguable use such a macro.

iabdalkader, 2021-08-11:
> `extmod/modussl_mbedtls.c:mbedtls_raise_error`

Was it done like that to avoid this specific issue (casting to mp_rom_error_text_t ) ?

jimmo, 2021-08-11:
I think `makeqstrdefs.py` should be updated to fail when it's not passed a literal string in order to prevent the strcmp matching not being optmised out -- i.e. ` MP_ERROR_TEXT(ffs_strerror(res))` shouldn't be allowed. I have a WIP fix, will send PR tomorrow.

I'm a little bit nervous about the proposed `MP_ERROR_TEXT_FROM_STR` suggestion, because `mp_raise_msg` assumes that the string is long-lived (e.g. static or on the heap). I'm assuming that `ffs_strerror` behaves like regular `strerror` and returns a static string, but it would be bad if someone passed in a stack-allocated buffer.

A safer alternative might be to use `mp_raise_msg_varg(&mp_type_OSError, MP_ERROR_TEXT("FFS error: %s"), ffs_strerror(res))`. Or even better of course, if possible, map `res` into one of the existing `MP_Exxx` values.

iabdalkader, 2021-08-11:
> I'm assuming that `ffs_strerror` behaves like regular `strerror` and returns a static string

Yes they all return pointer to static string, but wouldn't a stack buffer also work since mp_raise_xxx don't return ? 

> A safer alternative might be to use `mp_raise_msg_varg(&mp_type_OSError, MP_ERROR_TEXT("FFS error: %s"), ffs_strerror(res))`.

Yes this is a good solution, maybe add a macro for this case ?

```c
#define mp_raise_msg_from_str(exc_type, str) mp_raise_msg_varg(exc_type, MP_ERROR_TEXT("%s"), str)
....
mp_raise_msg_from_str(&mp_type_OSError, ffs_strerror(res))
```