micropython/micropython #7639: micropython dict has not __contains__ method

bhcuong2008, 2021-08-11:
Hi Team,

Today I check __contains__ method of dict, but not found. MicroPython v1.16.  I open file objdict.c, and also not found. Is it missed?

Thanks,
Cuong.

dlech, 2021-08-11:
Search for `MP_BINARY_OP_CONTAINS`.

bhcuong2008, 2021-08-11:
@dlech , Could I add this to dict_locals_dict_table of dict obj? Is it ok?

{ MP_ROM_QSTR(MP_QSTR___contains__), MP_ROM_PTR(&mp_op_contains_obj) }

bhcuong2008, 2021-08-11:
I've just added and built it. It's ok now. dict class now has __contains__ method.

dlech, 2021-08-11:
Why do you need this method? Usually people just use the `in` operator.

bhcuong2008, 2021-08-11:
@dlech , I create a class inherited from dict. It has __contains__ method. Inside this, I need to call super `__contains__` to do 'in', 'not in' operator.