micropython/micropython #7642: PyDFU and Mac M1

thestumbler, 2021-08-12:
I ran into a problem trying to use the **make ... deploy** command, kept getting "No backend available" error messages. Doing some searching around, I found references to this issue, for example [here](https://github.com/pyusb/pyusb/issues/355) and [here](https://gitmemory.com/issue/micropython/micropython/4933/518866430).

I kept thinking it was my installation, but couldn't pin it down. Finally out of desperation I tried the hack suggested by the fellow in the second link above, and put this at the top of **tools/pydfu.py**

```
import sys
if sys.platform == 'darwin':
    import os
    os.environ['DYLD_FALLBACK_LIBRARY_PATH'] = '/opt/homebrew/lib'
```

That did the trick. This clearly isn't a Micropython issue, but I thought I'd post this here in case it can help someone else in the same situation. I suspect the problem is related to pyenv or the homebrew installation thereof. 

System:

Mac mini M1 2020, Big Sur 11.5
Python 3.9.6 via pyenv 
( started off with 3.9.5 but same error )

dpgeorge, 2021-08-12:
It might be possible to do this at the shell before running `make deploy`:
```
$ export DYLD_FALLBACK_LIBRARY_PATH=/opt/homebrew/lib
```

thestumbler, 2021-08-13:
Alas, that doesn't do the trick. Just did a quick test. Oh well...

closed 2025-10-03 by jonnor