micropython/micropython #7653: How to handle optional compiler features in mpy-cross?

dlech, 2021-08-13:
This is something I've come across a few times before and looking at #7649 got me thinking about it again.

A single `mpy-cross` executable should be able to build `.mpy` files that run on any MicroPython port, including customized ones. However, not all ports support all features.

Examples:
- On a port without floating point support, running a .mpy file with floating points can cause a crash due to invalid CPU instruction. This is particularly subtle with the `/` operator and integers.
- There are some options, like `MICROPY_PY_BUILTINS_SET` that affect the compiler output but may be disabled in a port which will fail to process the bytecodes.
- #7649 is probably a little different in that the compiled .mpy would run fine since it basically transpiles the code to something else, but if the same .py source was copied to a device and tried to compile on a device without f-string support enabled, then compiling the same file would fail there.

So what I'm wondering is if it is reasonably possible to add more options to `mpy-cross` like `-mno-unicode` to basically make `mpy-cross` give the exact same compiler output as any given MicroPython port? I.e. `-mno-float` would cause `mpy-cross` to give a compiler error if floats were used anywhere instead of successfully compiling the program.

dpgeorge, 2021-08-14:
Related to this, and looking at it from the opposite point of view, I think it would be good if `mpy-cross` had *fewer* options, and .mpy files containing just bytecode (ie no native/machine code) should be able to run on all MicroPython targets.  I think it's quite easy to remove the `-mno-unicode` option.  `-mcache-lookup-bc` is harder but possible to eliminate if ports that have this caching option enabled expande bytecode loaded from .mpy so it has slots for the caching (also requires adjusting jump offsets).

dlech, 2021-08-18:
Fewer options makes sense from a maintenance and just keeping things simple point of view. So I guess the solution for catching things at compile time rather than at run time would be an external static analysis tool.

mattklapman, 2025-01-21:
I cannot get `mpy-cross` to allow use of the `-O` optimization to remove my `if __debug__:` statements.
This seems to be an issue as the code will run slower than JIT with `opt_level(1)`.

If I am not missing anything here, then to me this is a glaring feature miss.

Can anyone confirm this is currently the case as of 1.24?