micropython/micropython #7654: nimble: Indicating a characteristic will also send indications for other characteristics.

jimmo, 2021-08-13:
Related to the `send_update` arg added to `ble.gatts_write` in #7564.

The following code on a server (with a connected but unsubscribed client) results in the client receiving an indication for both characteristics.

```py
ble.gatts_write(first_char_handle, "value", True)  # (1)
ble.gatts_write(second_char_handle, "value", False)
ble.gatts_indicate(conn_handle, second_char_handle)   # (2)
```

I would expect:
 - (1) should not be sent to an unsubscribed client.
 - (2) should only trigger an indication for `second_char_handle` (regardless of subscription state).

It's not obvious to me that this is being caused by modbluetooth.c / modbluetooth_nimble.c, so might need to investigate in NimBLE (or confirm that this is the expected behaviour). I don't see this for notify, only indicate.

dpgeorge, 2021-08-14:
Might be worth checking the behaviour against BTstack.

jimmo, 2021-08-14:
As far as I can tell, bluekitchen doesn't implement subscription (which is why #7564 is NimBLE-only) -- we need to implement handling of writes to the CCCD ourselves.