micropython/micropython #7656: esp32: Native UARTs crashing the system

fred-cardoso, 2021-08-13:
Hi guys!

I am using microPython on a ESP32-DevKitC-VE and I was trying to setup the UART2 with the following code:

```
from machine import UART
uart1 = UART(1, baudrate=9600, tx=17, rx=16)
```

It crashed the system with the following message:

```
>>> from machine import UART
>>> uart1 = UART(1, baudrate=9600, tx=17, rx=16)
NotImplementedError: opcode
>>> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d7771  PS      : 0x00060730  A0      : 0x800d5f1b  A1      : 0x3ffcd2c0
A2      : 0x3ffc42f4  A3      : 0x00000000  A4      : 0x3ffcd2c8  A5      : 0x3ffbeebc
A6      : 0x0000000c  A7      : 0xff000000  A8      : 0x5ffcd2bc  A9      : 0xffffffff
A10     : 0x00000000  A11     : 0x80000000  A12     : 0xffffffff  A13     : 0x00000001
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000001  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000003  LBEG    : 0x400d7761  LEND    : 0x400d776d  LCOUNT  : 0x00000000

Backtrace:0x400d776e:0x3ffcd2c0 0x400d5f18:0x3ffcd2f0 0x400f00e9:0x3ffcd310 0x400f0925:0x3ffcd330 0x400d5d84:0x3ffcd370

ELF file SHA256: 5dbd69b893c882a9

Rebooting...
ets Jul 29 2019 12:21:46
```

Note: The `NotImplementedError: opcode` only appeared once!
Then I tried with the other UART1 (which is not supposed to work because it's connected to the SPIFlash, as with UART0 which is connected to the USB providing the REPL):

```
from machine import UART
uart1 = UART(1, baudrate=9600, tx=10, rx=9)
```

With the fowllowing result:

```
>>> from machine import UART
>>> uart1 = UART(1, baudrate=9600, tx=10, rx=9)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e2850  PS      : 0x00060b30  A0      : 0x800d96fd  A1      : 0x3ffcd0c0
A2      : 0x00000541  A3      : 0x3f4385c8  A4      : 0x00000009  A5      : 0xffffffff
A6      : 0xffffffff  A7      : 0x3ffc6498  A8      : 0x800e2850  A9      : 0x3ffcd0a0
A10     : 0xffffffdf  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x00000000
A14     : 0x3ffbf7f0  A15     : 0x3ffc6498  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0xffffffe0  LBEG    : 0x4009309e  LEND    : 0x400930a9  LCOUNT  : 0x00000000

Backtrace:0x400e284d:0x3ffcd0c0 0x400d96fa:0x3ffcd0e0 0x400dbd21:0x3ffcd110 0x400e2a4d:0x3ffcd130 0x400e4bfd:0x3ffcd150 0x400dc934:0x3ffcd1f0 0x400e2e45:0x3ffcd260 0x400e2e6e:0x3ffcd280 0x400f0647:0x3ffcd2a0 0x400f09a5:0x3ffcd330 0x400d5d84:0x3ffcd370

ELF file SHA256: 5dbd69b893c882a9

Rebooting...
ets Jul 29 2019 12:21:46
```

Finally, I tried with a non-native UART port and it worked flawlessly:

```
>>> from machine import UART
>>> uart1 = UART(1, baudrate=9600, tx=18, rx=5)
>>> uart1.readline()
b'$GPGGA,165822.087,,,,<0$GPGGA,165823.087,,,,,0,00,,,M,,M,,*7C\r\n'
>>> uart1.readline()
b'$GPGSA,A,1,,,,,,,,,,,,,,,*1E\r\n'
>>> uart1.readline()
b'$GPRMC,165823.087,V,,,,,0.00,0.00,110180,,,N*40\r\n'
```

Any idea why this is happening?

I tried flashing `esp32spiram-20210623-v1.16.bin` and then `esp32spiram-20210813-unstable-v1.16-201-g671f01230` with the same results.

Thanks!

robert-hh, 2021-08-13:
GPIO16 and 17 are used for Flash. You have to use other GPIO pins for the UART, like you did.

fred-cardoso, 2021-08-13:
> GPIO16 and 17 are used on some boards for Flash or SPIRAM. You have to use other GPIO pins for the UART, like you did.

Hi @robert-hh 

Thanks for the help.
I was following the PinOut diagram and they only state that the 6,7,8,9,10,11 GPIO pins are used.
[image]

So, basically, the device has 3 UARTs and none are usable?

robert-hh, 2021-08-13:
The device has 3 UARTs and all are usable, since TX and RX can be assigned to any free Pin. You do not have to use the default pins, like your own test shows. But there are restrictions. UART0 is used for REPL and connected to the USB/UART bridge at GPIO 1 and 3. GPIO32-39 are input only, so they cannot be used for TX.

robert-hh, 2021-08-13:
These topics have been discussed a lot in the Micropython Forum, which is the better place for it. The site here is about issues and design discussion of the firmware. In your case, the firmware is not the problem.

fred-cardoso, 2021-08-13:
Got it @robert-hh !

Thank for the help.

closed 2021-08-13 by fred-cardoso