micropython/micropython #7658: ESP32 some peripherals are read only using Viper or mem32

JohnnyGoldCoast, 2021-08-14:
For example, MCPWM registers can be read but not written to with Viper or mem32

```
from micropython import const
from machine import mem32

MCPWM0 = const(0x3FF5E000)
MCPWM1 = const(0x3FF6C000)

print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG (reset value = 0)
mem32[MCPWM0] = 0x55      # try to change PWM_CLK_PRESCALE
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG
```

dpgeorge, 2021-08-14:
What happens when you write the same code in C, does it work?  If so, what does the assembler output look like for that C code?  And which ESP32 chip is this run on?

JohnnyGoldCoast, 2021-08-14:
I've been able to configure the MCPWM using  the driver in Arduino but I don't know how to view assembler output. Reading the next register (PWM_TIMER0_CFG0_REG) returns 0xFF00 as expected which is reset value.

The above code is able to access some of the other peripherals for example the TIMG registers can be written to.

Not sure which chip but it's a ESP-WROOM-32 module with 4MB flash running V1.16

A WROVER module with 4MB flash/8MB SPIRAM produces the same result

jimmo, 2021-08-14:
> I've been able to configure the MCPWM using the driver in Arduino but I don't know how to view assembler output.

`xtensa-esp32-elf-objdump -D path/to/firmware.elf`

Then you can search for the function that does the register writes (likely an IDF function). You might need to follow a couple of layers of calls.

dpgeorge, 2021-08-16:
I had a look at this and it turns out the MCPWM peripheral needs to have its clock enabled and reset line cleared before its registers can be modified (this is very similar to a lot of other MCUs).  The code looks like:
```python
from micropython import const
from machine import mem32

DR_REG_DPORT_BASE = const(0x3FF00000)
DPORT_PERIP_CLK_EN_REG = const(DR_REG_DPORT_BASE + 0x0C0)
DPORT_PERIP_RST_EN_REG = const(DR_REG_DPORT_BASE + 0x0C4)
DPORT_PWM0_CLK_EN = const(1 << 17)
MCPWM0 = const(0x3FF5E000)
MCPWM1 = const(0x3FF6C000)

# Enable CLK and disable RST.
print(hex(mem32[DPORT_PERIP_CLK_EN_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN_REG] & 0xffffffff))
mem32[DPORT_PERIP_CLK_EN_REG] |= DPORT_PWM0_CLK_EN
mem32[DPORT_PERIP_RST_EN_REG] &= ~DPORT_PWM0_CLK_EN
print(hex(mem32[DPORT_PERIP_CLK_EN_REG] & 0xffffffff))
print(hex(mem32[DPORT_PERIP_RST_EN_REG] & 0xffffffff))

print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG (reset value = 0)
mem32[MCPWM0] = 0x55      # try to change PWM_CLK_PRESCALE
print(hex(mem32[MCPWM0])) # read PWM_CLK_CFG_REG
```
(tested on a TinyPICO)

JohnnyGoldCoast, 2021-08-18:
Thanks very much Damien for making the effort to look into this. I'll close this issue.

dpgeorge, 2021-08-19:
Might be good to provide some documentation about this because it's useful knowledge.

JohnnyGoldCoast, 2021-08-19:
The original post arose because I have a limited understanding of the ESP32 at the chip level.  Each of the peripheral modules needs to have its clock enabled and reset de-asserted as discovered by Damien. This is why some modules appeared to be R/W whilst others were read only.

Details can be found in section 5.3.7 Peripheral Clock Gating and Reset of the technical reference manual which can be found here:
[](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

At 721 pages it must be right up there in terms of the sheer volume of information it contains but it provides an insight into the capabilities of this remarkably versatile chip.

My efforts are directed at understanding the MCPWM module(s) at the register level for use in the prototyping of DC-DC power converters. Micropython will be extremely useful for running open loop tests where switching frequency, duty cycle and dead-times can easily be adjusted on the fly. Closing the loop to provide output voltage regulation may be possible using the MCPWM's capture submodule whilst the fault detection submodule could protect against over-current and over/under voltage conditions. Whether Micropython can execute the control loop rapidly and deterministically enough remains to be seen.

closed 2021-09-01 by JohnnyGoldCoast

dpgeorge, 2022-07-20:
The ESP32-S2 might have additional enable registers that need to be enabled first.  Or maybe the `SYSTEM_PERI_BUS_1` address is different, or the offset of the peripheral registers.  You'll need to read the ESP32-S2 datasheet (or dive into the IDF) to find that information.