micropython/micropython #7663: ESP32 SPI - request support for hardware-managed devices

oclyke, 2021-08-14:
The ESP32 SPI peripherals support limited management of attached devices in their abstraction layer (I am not sure whether this is a software feature of their HAL or a hardware capability of the actual peripheral though).

The hardware SPI interface common to microcontroller boards in MicroPython does not support this capability - and neither should it. However it would be advantageous to add an additional API which allows ESP32 users to take advantage of these features for both speed and convenience.

Additional capabilities:
* hardware management of CS lines
* independent clock frequency settings
* independent phase and polarity settings

Proposed API:
``` python
# import from esp32 module because these features are ESP32 specific
from esp32 import SPIHost, SPIDev

# construct a SPIHost for the peripheral you want to use.
# similar to machine.SPI syntax except that device-specific properties are removed
vspi = SPIHost(2, sck=Pin(18), mosi=Pin(23), miso=Pin(19))

# add devices to the host with independent settings
# optionally allow HAL-controlled cs pin
dev1 = SPIDev(vspi, baudrate=80000000, cs=Pin(25), polarity=0, phase=0, bits=8, firstbit=0)
dev2 = SPIDev(vspi, baudrate=5000000, bits=8)

# note: the esp-idf drivers allow many (untested total) spi devices, but only 3 per host may 
# use a hw controlled cs pin. one may use a Pin to select additional spi devices

# do transactions on devices
devn.write(b'12345')     # write 5 bytes on MOSI
devn.read(10)            # read 10 bytes on MISO
```

This is also a different solution to #6340 (as opposed to calling ```spi.init(baudrate)``` to reconfigure)

I'd be willing to make a PR for this. Please let me know thoughts.

chrismas9, 2021-08-15:
Managed CS won't work for all devices, especially some ADCs. It's quite common for ADCs to require a single CS around an SPI write of config data, eg channel select, and SPI read of data. Not all ADCs use pure SPI standard. Some are Microwire, others proprietary serial, some put RDY or conversion complete on SDO, some use SCLK as the conversion clock and require additional clocks. More than half the devices I checked need continuous CS.

oclyke, 2021-08-20:
@chrismas9 good point - a default argument for ```cs=None``` would preserve the existing API functionality and allow users to implement their own CS management with a machine ```Pin``` of their own.

oclyke, 2021-08-20:
Thinking about this some more... Of course it would be possible to duplicate most of the existing machine SPI code from the ESP23 and simply make the desired changes, but this would increase code size and not benefit other ports down the line. 

Perhaps it is possible to add this API in a backward-compatible manner. See below proposition:
```
# import is now from machine module
from machine import SPI, SPIDev

# the SPI class is now an API to create a SPIDev as the sole device on the selected SPI host peripheral
# the returned device is not given a hardware controlled cs pin
# inherit SPIDev defaults but still allow specification of device-level settings
# it is now possible to specify a hardware cs pin from the SPI constructor if desired
vspi = SPI(2, sck=Pin(18), mosi=Pin(23), miso=Pin(19))
hspi = SPI(1, sck=Pin(14), mosi=Pin(13), miso=Pin(12), cs=Pin(25), polarity=0, phase=0, bits=8, firstbit=0)

# add devices to the host with independent settings
# optionally allow HAL-controlled cs pin
dev1 = SPIDev(vspi, baudrate=80000000, cs=Pin(25), polarity=0, phase=0, bits=8, firstbit=0)
dev2 = SPIDev(vspi, baudrate=5000000, bits=8)

# note: the esp-idf drivers allow many (untested total) spi devices, but only 3 per host may 
# use a hw controlled cs pin. one may use a Pin to select additional spi devices

# do transactions on devices
devn.write(b'12345') 	# write 5 bytes on MOSI
devn.read(10)        	# read 10 bytes on MISO

# use the SPI 'host' object as a device (which it is)
vspi.write(b'12345')	# write 5 bytes on vspi device MOSI
hspi.read(10)			# read 10 bytes on hspi device MISO
```

Doing this would require:
* ports to emit exceptions when user tries to configure ```cs``` keyword in SPI constructor (if not supported)
* addition of the SPIDev class which can be 'linked' to a hardware SPI host instance
* re-factor of SPI constructor to return SPIDev object which is already linked to desired SPI host instance

Doing so would allow:
* ports to adopt a per-device settings model for phase, polarity, baud rate, and hardware cs control
* backward compatibility with existing API

oclyke, 2021-08-20:
@dpgeorge I'm debating whether this would be an ESP32 or general microcontroller PR. Do you have any thoughts about this? If going general we could edit the title and label. Thanks for looking!

oclyke, 2021-08-20:
Okay, this seems like it may be two different feature requests.

1) support for 'automatic' or 'managed' cs lines in spi
2) support for 'devices' which have independent settings sharing one spi bus 

The second could build on the first, but the first is independent of the second. 

#7693 is a PR that implements the managed cs lines in SoftSPI and esp32 port hardware SPI.

dpgeorge, 2022-06-21:
As mentioned in #7693, I think it'd be better to make a higher-level `SPIDevice` class, with a context manager.