micropython/micropython #7664: `uasyncio.rst` and `uctypes.rst` are still named with `u` prefix

hlovatt, 2021-08-15:
In the documents library files `uasyncio.rst` and `uctypes.rst` are still named with `u` prefix. The other similar files have `u` removed, e.g. `os.rst`.

jimmo, 2021-08-15:
This is intentional, although I agree slightly confusing.

uasyncio is because it's written in Python (and usually frozen), rather than being built-in (i.e. written in C). The actual library is called uasyncio. There is (currently) no mechanism to automatically convert `ufoo` to `foo` for Python modules (either on the filesystem or frozen).

(Historically it was called uasyncio because it was API incompatible with Python's asyncio. Since uasyncio v3 it is now compatible with Python's, so it might be worth making the case for renaming it).

uctypes has very little overlap with Python ctypes. It really is a "micro ctypes". I suggested in #5377 that maybe it should be renamed "ntypes" or "native" or something to really make it clear that it's different to Python's "ctypes". (While leaving behind an alias for uctypes).

jimmo, 2021-08-15:
I think it would be worth clarifying both of these in their respective pages though, so thanks for pointing this out @hlovatt !

hlovatt, 2021-08-15:
Thanks for your explanation. I think your suggestions of changing to `asyncio` and `ntypes` or `native` are good ideas. Current naming not obvious to me!