micropython/micropython #7665: Unix build errors

peterhinch, 2021-08-15:
After getting the latest source, the Unix build fails (others e.g. Pyboard D still work).
```
$ cd ports/unix/
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/unix
$ make -j8
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-standard/genhdr/qstr.i.last
In file included from ../../lib/axtls/ssl/asn1.c:39:
../../lib/axtls/ssl/os_port.h:78:10: fatal error: ../../../extmod/crypto-algorithms/sha256.h: No such file or directory
   78 | #include "../../../extmod/crypto-algorithms/sha256.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Followed by further messages with the same complaint. There is no `extmod/crypto-algorithms/` directory on my clone, although there is a file `extmod/extmod.cmake` - this doesn't seem to point to such a directory.
```bash
$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```

jimmo, 2021-08-15:
@peterhinch This is because the axtls submodule has been updated (in 75c1609c3a36f40f2dc9d5f92a8b35678eb6e6d8), and the usual trick of just doing `make submodules` (or manually `git submodule update`) doesn't work because the remote has changed (to https://github.com/micropython/axtls).

It should be sufficient to run

```
git submodule sync lib/axtls
cd ports/unix
make submodules
```

@dpgeorge should we consider adding `git submodule sync` to the `submodules` rule in mkrules.mk?

From #7666 it looks like you need to do this for pico-sdk and tinyusb too. If this doesn't work then the nuclear option is to `git submodule deinit -f lib/foo` followed by `rm -rf .git/modules/lib/foo`. 

To fix the PR... I think this should work:

```
git checkout framebuf-palette
git reset origin/master lib/axtls lib/pico-sdk lib/tinyusb   # restore the submodules to the previous commit (assuming origin is micropython/micropython, change to whatever you use as the upstream remote)
git submodule sync lib/axtls lib/pico-sdk lib/tinyusb   # fix submodule remotes
git submodule update --init lib/axtls lib/pico-sdk lib/tinyusb   # update submodules
git add lib/axtls lib/pico-sdk lib/tinyusb   # add the "reversion" to the commit
git commit --amend   # append to the commit, this should cancel out the changes in the commit. while it's prompting to edit the commit message, the commit should now just have your modified files
git status  # the working dir should be clean
git push -f   # assuming the upstream was set, otherwise add <your remote> <framebuf-palette>
```

peterhinch, 2021-08-15:
I'm afraid this had no effect on the Unix build: still getting the exact same errors.

dpgeorge, 2021-08-16:
> should we consider adding `git submodule sync` to the `submodules` rule in mkrules.mk?

Yes that sounds good.

> I'm afraid this had no effect on the Unix build: still getting the exact same errors.

What do you get when you run `cat .git/modules/lib/axtls/config`?  It should contain:
```
[remote "origin"]
	url = https://github.com/micropython/axtls.git
	fetch = +refs/heads/*:refs/remotes/origin/*
```

peterhinch, 2021-08-16:
Yes. This is the complete text:
```bash
$ cat .git/modules/lib/axtls/config
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	worktree = ../../../../lib/axtls
[remote "origin"]
	url = https://github.com/micropython/axtls.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
```

dpgeorge, 2021-08-16:
Try `git submodule deinit lib/axtls` and then `git submodule update --init lib/axtls`

peterhinch, 2021-08-16:
Sorry but still no dice (cut to restrict to first error):
```bash
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
$ git submodule deinit lib/axtls
Cleared directory 'lib/axtls'
Submodule 'lib/axtls' (https://github.com/micropython/axtls.git) unregistered for path 'lib/axtls'
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
$ git submodule update --init lib/axtls
Submodule 'lib/axtls' (https://github.com/micropython/axtls.git) registered for path 'lib/axtls'
Submodule path 'lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
$ cd ports/unix/
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/unix
$ make -j8
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-standard/genhdr/mpversion.h
GEN build-standard/genhdr/qstr.i.last
In file included from ../../lib/axtls/ssl/asn1.c:39:
../../lib/axtls/ssl/os_port.h:78:10: fatal error: ../../../extmod/crypto-algorithms/sha256.h: No such file or directory
   78 | #include "../../../extmod/crypto-algorithms/sha256.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

dpgeorge, 2021-08-16:
Please do:
```
$ cd lib/axtls
$ git log
$ cat ssl/os_port.h
```

peterhinch, 2021-08-16:
OK, I assume the top of git log will suffice:
```bash
$ cd lib/axtls
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/lib/axtls
$ git log
commit 43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0 (HEAD)
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Nov 5 11:35:51 2017 +0200

    crypto/aes.c: AES_set_key: If "iv" is NULL, don't copy it.
    
    IV is not needed for ECB cipher mode.

commit a1a6151c528c40920014783386185b5f46574229
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Nov 5 11:32:59 2017 +0200

    crypto/aes.c: Make public AES_encrypt(), AES_decrypt().
    
    Currently, axTLS exposes only CBC mode functions, the above functions
    may be useful to implement ECB mode. (Note that the above functions
    expect data in big endian order.)
    
    Motivatation: Implement MicroPython's "ucrypto" module, reusing
    available axTLS routines.

commit dac9176cac58cc5e49669a9a4d404a6f6dd7cc10
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Oct 29 15:54:22 2017 +0300

    ssl/tls1.c: If underlying stream returned EAGAIN, return SSL_EAGAIN.
    
    This helps better implement blocking vs non-blocking streams on top of
    axTLS (e.g. in MicroPython). Value of SSL_EAGAIN is -4, care was taken
    to make sure there's no TLS alert with value of 4 (because axTLS returns
    negated alert values as errors).

commit 9b3092eb3b4b230a63c0c389bfbd3c55682c620f
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Tue Jun 13 17:30:02 2017 +0300

    ssl/tls1.c: Hack: dump record size on hitting SSL_ERROR_RECORD_OVERFLOW.
    
    This is MicroPython-specific hack to allow to easier understand why TLS
    conenction doesn't work with some sites on low-heap systems (where TLS
    record buffer size is set to far less than default 16K).

commit cf3c04293b9f21799ecb9328abcb2ed849984a69
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/lib/axtls
$ 
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/lib/axtls
$ cat ssl/os_port.h
/*
 * Copyright (c) 2007-2015, Cameron Rich
 * 
 * All rights reserved.
 * 
 * Redistribution and use in source and binary forms, with or without 
 * modification, are permitted provided that the following conditions are met:
 *
 * * Redistributions of source code must retain the above copyright notice, 
 *   this list of conditions and the following disclaimer.
 * * Redistributions in binary form must reproduce the above copyright notice, 
 *   this list of conditions and the following disclaimer in the documentation 
 *   and/or other materials provided with the distribution.
 * * Neither the name of the axTLS project nor the names of its contributors 
 *   may be used to endorse or promote products derived from this software 
 *   without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
 * A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR
 * CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
 * EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO,
 * PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
 * PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF
 * LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
 * NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 * SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */

/**
 * @file os_port.h
 *
 * Some stuff to minimise the differences between windows and linux/unix
 */

#ifndef HEADER_OS_PORT_H
#define HEADER_OS_PORT_H

#include "os_int.h"
#include <errno.h>
#include <sys/types.h>
#ifndef __ets__
#include <arpa/inet.h>
#endif
#include <sys/time.h>
#include "config.h"

ssize_t mp_stream_posix_write(void *sock_obj, const void *buf, size_t len);
ssize_t mp_stream_posix_read(void *sock_obj, void *buf, size_t len);
extern int mp_stream_errno;

#if 1
#define SOCKET_READ(A,B,C)      mp_stream_posix_read((void*)A,B,C)
#define SOCKET_WRITE(A,B,C)     mp_stream_posix_write((void*)A,B,C)
#define SOCKET_CLOSE(A)         NOT_USED_IN_LIB_CODE
#define SOCKET_ERRNO() mp_stream_errno
#else
#define SOCKET_READ(A,B,C)      read(A,B,C)
#define SOCKET_WRITE(A,B,C)     write(A,B,C)
#define SOCKET_CLOSE(A)         if (A >= 0) close(A)
#define SOCKET_ERRNO()          errno
#endif
#define ax_calloc(x, y) calloc(x, y)
#define ax_open(x, y) open(x, y)

#ifndef be64toh
#define be64toh(x) __be64_to_cpu(x)
#endif

#define SSL_CTX_MUTEX_INIT(A)
#define SSL_CTX_MUTEX_DESTROY(A)
#define SSL_CTX_LOCK(A)
#define SSL_CTX_UNLOCK(A)

#define TTY_FLUSH()

#include "../../../extmod/crypto-algorithms/sha256.h"

#define SHA256_CTX CRYAL_SHA256_CTX
#define SHA256_Init(a) sha256_init(a)
#define SHA256_Update(a, b, c) sha256_update(a, b, c)
#define SHA256_Final(a, b) sha256_final(b, a)

#endif
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/lib/axtls
$ 
```

jimmo, 2021-08-16:
@peterhinch So it looks like the submodule has the correct remote but is pointing to an old commit. Can you get a log for the submodule, i.e. `git log lib/axtls`.

```
git log lib/axtls
```

should look like:

```
commit 75c1609c3a36f40f2dc9d5f92a8b35678eb6e6d8
Author: Damien George <damien@micropython.org>
Date:   Thu Jul 8 23:40:33 2021 +1000

    lib/axtls: Update to latest axtls 2.1.5 wih additional commits.
    
    Changes are:
    - update axTLS from 2.1.3 to 2.1.5
    - os_port.h is now provided by the user of the library
    - PLATFORM_RNG_U8 can be defined for get_random
    - fix -fsanitize=undefined diagnostics
    
    Signed-off-by: Damien George <damien@micropython.org>

commit 99bf5448bd02f9b97f55025725e7d2d4ba9aced8
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Nov 5 11:37:05 2017 +0200

    axtls: Update, exposes AES functions to implement ECB chiper mode.

commit 8f9af63c20fbdb49a0e9a29248b8c7abe6f01b61
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Oct 29 19:52:56 2017 +0200

    lib/axtls: Update, support for SSL_EAGAIN return code.
    
    A step towards implementing non-blocking stream support for SSL.

commit f01c1c6b35a94d717c2174662fb242bf235f6978
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Tue Jun 13 17:39:20 2017 +0300

    lib/axtls: Upgrade to axTLS 2.1.3 + MicroPython patchset.
    
    axTLS 2.1.3 brings support for TLS 1.2 and SNI. With MicroPython patchset
    on top of it, the code size growth (x86) is ~2K.
```

If it doesn't match, then I think your branch has reverted the submodule (which I think is the case in your `framebuf-palette` branch).

Can you do a `git log` on the micropython repo itself too?

dpgeorge, 2021-08-16:
> So it looks like the submodule has the correct remote but is pointing to an old commit. Can you get a log for the submodule, i.e. `git log lib/axtls`.

That the log shown above.

And further above we can see that it is indeed the correct remote but pointing to this old commit.
```
$ git submodule update --init lib/axtls
Submodule 'lib/axtls' (https://github.com/micropython/axtls.git) registered for path 'lib/axtls'
Submodule path 'lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
```

So @peterhinch what does `git status` say (in the top level dir of the micropython repo), and also `git log` (at the same location)?

dpgeorge, 2021-08-16:
> > So it looks like the submodule has the correct remote but is pointing to an old commit. Can you get a log for the submodule, i.e. `git log lib/axtls`.
> 
> That the log shown above.

That's not right, my bad.  These things are different: `cd lib/axtls && git log` vs `git log lib/axtls`.

peterhinch, 2021-08-16:
Hell, if it's confusing you two, think what it's doing to me :-)
```
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
$ git status
On branch master
Your branch is ahead of 'origin/master' by 1 commit.
  (use "git push" to publish your local commits)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   lib/btstack (untracked content)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	examples/natmod/framebuf_utils/
	examples/natmod/pete/
	ports/esp8266/modules/typescript
	tools/mpr

It took 2.15 seconds to enumerate untracked files. 'status -uno'
may speed it up, but you have to be careful not to forget to add
new files yourself (see 'git help status').
no changes added to commit (use "git add" and/or "git commit -a")
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
```
```
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
$ git log
commit cbb4dc317fe658e477ebf1144dfd0f5951b24578 (HEAD -> master, peterhinch/master)
Author: Peter Hinch <peter@hinch.me.uk>
Date:   Sat Aug 14 17:42:35 2021 +0100

    Test commit.

commit 5733c49174a200b59196f44377a1418a6c38c877 (origin/master, origin/HEAD)
Author: Jim Mussared <jim.mussared@gmail.com>
Date:   Thu Jul 16 13:07:41 2020 +1000

    stm32,esp32,rp2: Enable MICROPY_PY_FSTRINGS by default.
    
    Signed-off-by: Jim Mussared <jim.mussared@gmail.com>

commit 692d36d779192f32371f7f9daa845b566f26968d
Author: Jim Mussared <jim.mussared@gmail.com>
Date:   Fri Aug 13 01:44:08 2021 +1000

    py: Implement partial PEP-498 (f-string) support.
    
    This implements (most of) the PEP-498 spec for f-strings and is based on
    https://github.com/micropython/micropython/pull/4998 by @klardotsh.
    
    It is implemented in the lexer as a syntax translation to `str.format`:
      f"{a}" --> "{}".format(a)
    
    It also supports:
      f"{a=}" --> "a={}".format(a)
    
    This is done by extracting the arguments into a temporary vstr buffer,
    then after the string has been tokenized, the lexer input queue is saved
    and the contents of the temporary vstr buffer are injected into the lexer
    instead.
    
    There are four main limitations:
    - raw f-strings (`fr` or `rf` prefixes) are not supported and will raise
      `SyntaxError: raw f-strings are not supported`.
    
    - literal concatenation of f-strings with adjacent strings will fail
        "{}" f"{a}" --> "{}{}".format(a)    (str.format will incorrectly use
[... 89 lines cut ...]
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Nov 5 11:37:05 2017 +0200

    axtls: Update, exposes AES functions to implement ECB chiper mode.

commit 8f9af63c20fbdb49a0e9a29248b8c7abe6f01b61
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Sun Oct 29 19:52:56 2017 +0200

    lib/axtls: Update, support for SSL_EAGAIN return code.
    
    A step towards implementing non-blocking stream support for SSL.

commit f01c1c6b35a94d717c2174662fb242bf235f6978
Author: Paul Sokolovsky <pfalcon@users.sourceforge.net>
Date:   Tue Jun 13 17:39:20 2017 +0300

    lib/axtls: Upgrade to axTLS 2.1.3 + MicroPython patchset.
    
    axTLS 2.1.3 brings support for TLS 1.2 and SNI. With MicroPython patchset
    on top of it, the code size growth (x86) is ~2K.

commit 26d5e91bf39463ca72f0a5f5d2c79b8be1d29541
:
```

jimmo, 2021-08-16:
@peterhinch 

> commit cbb4dc317fe658e477ebf1144dfd0f5951b24578 (HEAD -> master, peterhinch/master)

This commit reverts the submodule (https://github.com/peterhinch/micropython/commit/cbb4dc317fe658e477ebf1144dfd0f5951b24578), and that's what's preventing it updating. I would guess that if you drop this commit then `git submodule update lib/axtls` you'll be right.

Lots of options but I'd probably go with

```
git fetch origin
git reset --hard origin/master
git submodule update --init lib/axtls
```

(Assuming origin is `micropython/micropython`, which it looks like it is)

That won't touch your untracked files but will bring your local `master` to `origin/master`.

peterhinch, 2021-08-16:
@jimmo  You are a star! That has fixed it.

To prevent this happening again, should I mod my build script? Bear in mind that where Git is concerned my expertise doesn't extend much beyond following instructions. Currently I periodically run an update script which does the following:
```bash
cd $MPDIR
echo Working...
git checkout master
git pull origin master --tags
git submodule update --init
cd mpy-cross
make clean
make -j 8
cd ../ports/stm32
make BOARD=PYBV11 clean
make BOARD=PYBV10 clean
make BOARD=PYBLITEV10 clean
make BOARD=PYBD_SF2 clean
make BOARD=PYBD_SF3 clean
make BOARD=PYBD_SF6 clean
cd ../esp8266
make clean
cd ../unix
make clean
make -j 8
```
Should I be doing more e.g. `git submodule update --init --recursive` ? Or some other thaumaturgy?

Regarding my PR, after all this will your above instructions still work or should I create a new branch and PR?

jimmo, 2021-08-17:
> Currently I periodically run an update script which does the following:

I _think_ adding `git submodule sync` before the `git submodule update` would have solved it. But perhaps the better solution would be to change the script to use `make submodules` in each of the ports (instead of the `git submodule update`), and fix the submodules target to also do a sync (as discussed above). Sounds like @dpgeorge was on board, I will send a PR.

peterhinch, 2021-08-17:
@jimmo I'm still having difficulties with my PR. Following your original instructions I get:
```
$ git checkout framebuf-palette
M	lib/axtls
M	lib/btstack
Switched to branch 'framebuf-palette'
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/unix
$ git reset origin/master lib/axtls lib/pico-sdk lib/tinyusb
Unstaged changes after reset:
M	lib/axtls
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/unix
$ git submodule sync lib/axtls lib/pico-sdk lib/tinyusb
error: pathspec 'lib/axtls' did not match any file(s) known to git
error: pathspec 'lib/pico-sdk' did not match any file(s) known to git
error: pathspec 'lib/tinyusb' did not match any file(s) known to git
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/unix
$ 
```
I have not made any changes to axtls, pico-sdk, tinyusb or btstack so I'm puzzled as to why they come up with "M". So these crop up in my commits even if I create a new branch in an attempt to start from scratch (which I'm happy to do if it's the best solution and can be made to work).

If I made any changes in the master branch it was in error and I'm happy to discard them. Can the master branch be forced to match upstream?

jimmo, 2021-08-17:
```
$ git submodule sync lib/axtls lib/pico-sdk lib/tinyusb
error: pathspec 'lib/axtls' did not match any file(s) known to git
error: pathspec 'lib/pico-sdk' did not match any file(s) known to git
error: pathspec 'lib/tinyusb' did not match any file(s) known to git
```

Needs to run in the top-level, not ports/unix.

peterhinch, 2021-08-17:
Your git commands now ran. However I had these results:
```
$ git commit --amend
[framebuf-palette 793f90283] extmod/modframebuf: Enable blit between different formats.
 Date: Sun Aug 15 09:57:38 2021 +0100
 7 files changed, 63 insertions(+), 9 deletions(-)
 create mode 100644 tests/extmod/framebuf_palette.py
 create mode 100644 tests/extmod/framebuf_palette.py.exp
```
```
$ git status
On branch framebuf-palette
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   lib/btstack (untracked content)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	examples/natmod/framebuf_utils/
	examples/natmod/pete/
	ports/esp8266/modules/typescript
	tools/mpr

no changes added to commit (use "git add" and/or "git commit -a")
```
The untracked files in examples/natmod are mine. The rest are not - where have these come from? 

I went ahead with the force push, but looking with gitk I'm not at all confident that the cruft has been lost. It's undoubtedly a case of old dogs and new tricks but this git stuff has me completely baffled. Sorry to be a nuisance. :(

jimmo, 2021-08-17:
> The rest are not - where have these come from?

- lib/btstack -- you probably don't need this, so safe to just do `git submodule deinit -f lib/btstack` which will remove the contents of lib/btstack.  (If you later run `make submodules` in ports/stm32, it'll re-add it, but it'll be a clean clone). Alternatively, you can `cd lib/btstack` and `git status` will tell you what needs to be done.

 - ports/esp8266/modules/typescript -- I'm not sure... as it's untracked, it's probably just safe to delete.

 - tools/mpr -- possibly at some stage you checked out a WIP branch of mpremote (and I _think_ at some stage it was called `mpr`). Also probably safe to delete, but maybe check what it is first (if you have micropython/tools in your $PATH, then `mpr` could be a wrapper for mpremote?). (FWIW, I have `alias mpr='/home/jimmo/src/github.com/micropython/micropython/tools/mpremote/mpremote.py'` in my .bash_aliases).

I feel like the way git handles submodules leads to a lot of subtle traps, especially if anything in the submodule changes. For example, I don't understand why `submodule update` doesn't automatically do `submodule sync`, and I also spend a non-trivial amount of time ensuring my submodules are in the right state. Unfortunately, I'm not aware that there's a better way to handle what we do with MicroPython's dependencies.

Looking at https://github.com/peterhinch/micropython/commits/framebuf-palette it still appears to have the submodule changes.

peterhinch, 2021-08-17:
That explanation of the untracked files makes sense and I have done as you suggest. So this now looks right:
```
$ git status
On branch framebuf-palette
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	examples/natmod/framebuf_utils/
	examples/natmod/pete/

nothing added to commit but untracked files present (use "git add" to track)
```
What should I do to fix the PR?

peterhinch, 2021-08-18:
While my above script can grab the latest sources, my master branch still looks like this:
```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   lib/pico-sdk (modified content)
	modified:   lib/tinyusb (modified content, untracked content)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	examples/natmod/framebuf_utils/
	examples/natmod/pete/
	tests/extmod/framebuf_palette.py
	tests/extmod/framebuf_palette.py.exp

no changes added to commit (use "git add" and/or "git commit -a")
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython
```
The untracked files make sense but the modified files mean that even if I create a new branch for my PR, the commit will include this cruft. I'm foxed as to how to deal with this.

davehylands, 2021-08-18:
Assuming that you have no actual modified content that you want to keep, you should be able to follow these steps to clean things up.

I see that lib/axtls used to point to pfalcon/axtls and now it points to micropython/axtls.

For pico-sdk and tinyusb, I was able to do:

```
rm -rf lib/pico-sdk lib/tinyusb
git submodule update --init
```
For axtls, I had to get a little more creative.
I edited `.git/config` and changed `https://github.com/pfalcon/axtls` to `https://github.com/micropython/axtls` and then did:
```
rm -rf .git/modules/lib/axtls
rm -rf lib/axtls
git submodule update --init
```

peterhinch, 2021-08-18:
Thanks, Dave, I think that's fixed it. I've managed to create a new branch and PR without all the cruft.

closed 2021-08-18 by peterhinch

peterhinch, 2021-08-19:
Now sorted, thanks @dpgeorge, @jimmo, @dhylands