micropython/micropython #7687: genhdr/moduledefs.h missing in build dir for stm32 (Windows build)

JohnieBraaf, 2021-08-19:
I'm building micropython stm32 ports on Windows 10 directly.
This works fine for the default board (PYBV10), however for other boards like PYBD_SF6 it fails with:

`../../py/objmodule.c:36:10: fatal error: genhdr/moduledefs.h: No such file or directory
   36 | #include "genhdr/moduledefs.h"`

The file is indeed missing in the build directory. If I copy it over from the build-PYBV10, the build succeeds.
On Linux machines this problem does not seem to occur.

Any pointers on why this file is omitted in the build preparation step?

jimmo, 2021-08-19:
genhdr/moduledefs.h is generated during the build (it's searching for all .c files that define a built-in module and generating the registration code for them).

My guess is that this is the same as https://github.com/micropython/micropython/issues/7012 .... we've just gone over some limit on Windows for the maximum number of command line arguments....

JohnieBraaf, 2021-08-19:
I think you are right, the collection of files passed to makemodulefs.py is getting truncated.
Also I'm afraid that GEN build-xxx/genhdr/qstr.i.last will be subject to similar issues.

JohnieBraaf, 2021-09-17:
I have found that [MSYS2](https://www.msys2.org/) can build very nicely on windows withouth this limitation.
Therefore I'm closing this issue.

closed 2021-09-17 by JohnieBraaf