micropython/micropython #7699: ESP32:How to use micopython ethernet to ap?How to route ethnet to ap?

OceanSeek, 2021-08-24:
import network
import time

from machine import Pin
lan = network.LAN(mdc=Pin(16), mdio=Pin(17), power=None, id=None, phy_addr=1, phy_type=network.PHY_LAN8720)
try:
    lan.active(True)
except Exception as err:
    print("err:", err)
ret = lan.ifconfig()
print(ret)

ap = network.WLAN(network.AP_IF) # create access-point interface
ap.config(essid='ESP-AP') # set the ESSID of the access point
ap.config(max_clients=10) # set how many clients can connect to the network
ap.active(True)         # activate the interface
ap.ifconfig()
print("ap ip:", ap.ifconfig())

lan an ap can get the right ip,. Lan can connect to the ethernet,but my phone conncet the ap can not conncet to ethernet.

dpgeorge, 2021-08-24:
There is currently no support to route between network interfaces.

See related #5369.

beyonlo, 2021-08-24:
> There is currently no support to route between network interfaces.
> 
> See related #5369.

@dpgeorge  I think that with IP_FORWARD=1 this problem is solved, and many others problems/scenarios will solved too - I would like so much too, to forward between AP and STA, or like this issue, between LAN and WLAN.  

Perhaps the last IDF version allows this configuration? If yes, would be wonderful micropython has a config to set up that, like this:

esp32.ip_forward = 1
esp32.ip_forward = 0

Default can be disable (0), but the important is user option to chance that.

Thank you very much.

dpgeorge, 2021-08-25:
> Perhaps the last IDF version allows this configuration? If yes, would be wonderful micropython has a config to set up that, like this:
> 
> esp32.ip_forward = 1
> esp32.ip_forward = 0
> 
> Default can be disable (0), but the important is user option to change that.

Unfortunately it looks like `CONFIG_LWIP_IP_FORWARD` needs to be a static value (either 0 or 1) and can't be dynamically changed at runtime.