micropython/micropython #7726: WEBREPL on FeatherS2 error

Maddox-zephyr, 2021-08-28:
I flashed the firmware featherS2-20210825-unstable-v1.16-237-g996f70316.bin.

WIFI sta connects and I can establish socket connections normally. I setup webrepl and it completes normally.
`>>> import webrepl_setup
WebREPL daemon auto-start status: enabled

Would you like to (E)nable or (D)isable it running on boot?
(Empty line to quit)
> E
To enable WebREPL, you must set password for it
New password (4-9 chars): xxxxx
Confirm password: xxxxxx
No further action required
`
When I go to use webrepl and hit connect, it just returns "disconnected."  The serial webrepl has the following:
`WebREPL connection from: ('192.168.1.149', 64425)
dupterm: EOF received, deactivating`

I also built the image with IDF 4.3 and get identical reults :-(

Let me know if I can provide any additional info or if there is already a solution.

Cheers,

Bruce

dpgeorge, 2021-09-02:
I can reproduce the issue with a Feather S2 board.

But WebREPL does work using the CLI interface from https://github.com/micropython/webrepl.  Eg:
```
$ ./webrepl_cli.py repl 192.168.20.29 
Password: 
op:repl, host:192.168.20.29, port:8266, passwd:x.
Remote WebREPL version: (1, 16, 0)
1/2
0.5
>>> 5+6
11
```
So it's something to do with the browser WebREPL code with S2 (I tested with Firefox and Chromium, and both have the same behaviour).

Maddox-zephyr, 2021-09-03:
Thanks for looking into this. I also tried Microsoft Edge browser with same error resulting.

I can get and put files using webrepl_cli.py!  Monitoring the serial repl port, it always reports:
```
WebREPL connection from: ('192.168.1.116', 36946)
dupterm: EOF received, deactivating
```
even though the transfers succeed.  I can live with that, but I thought the error might be useful info.

Your example looks like it is sending "repl" file and then it executes repl.  When I try the same command, I get an error that repl does not exist, which indeed I don't have a file named "repl" in my local execution directory.  Where can I get this file so I can use the remote repl per your example?  I searched aroung the repo, but couldn't find it.
```
$ ./webrepl_cli.py **repl** 192.168.20.29 
Password: 
op:repl, host:192.168.20.29, port:8266, passwd:x.
Remote WebREPL version: (1, 16, 0)
1/2
0.5
>>> 5+6
11
```

dpgeorge, 2021-09-03:
> even though the transfers succeed. I can live with that, but I thought the error might be useful info.

As far as I understand, that "dupterm: EOF received, deactivating" is normal behaviour, just telling you that the connection was closed successfully.

> Your example looks like it is sending "repl" file and then it executes repl.

Ah, oops, that's my local branch!  Now pushed as a PR here: https://github.com/micropython/webrepl/pull/66

Wind-stormger, 2021-11-18:
Hello,I seem to have encountered similar problems, but I have no idea to solve them.

The board is ESP32-S2-saola ,firmware information is `MicroPython v1.17-76-gc42c1c871-dirty on 2021-10-13; ESP32-S2-WROVER with ESP32-S2`

I ran this code once in repl and completed `webrepl_setup`:

```python
import webrepl_setup
```

I put this code into `boot.py`:

```python
import network
import time
import webrepl
ssid='BPI'
password='12345678'
def do_connect():
    import network
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    if not wlan.isconnected():
        print('connecting to network...')
        wlan.connect(ssid, password)
        while not wlan.isconnected():
            pass
    print('network config:', wlan.ifconfig())

do_connect()

webrepl.start()
```

Expected:

```
connecting to network...
network config: ('192.168.x.x, '255.255.255.0', '192.168.x.x', '192.168.x.x')
WebREPL daemon started on ws://192.168.x.x:8266
Started webrepl in normal mode
```

So far, everything seems to be fine.

But when I open webrepl in the Edge browser, enter the address, and click Connect, it only return `Disconnected`.

I've tried [http://micropython.org/webrepl](http://micropython.org/webrepl) And has made sure to connect to HTTP, also tried [https://github.com/micropython/webrepl](https://github.com/micropython/webrepl)  `webrepl.html` .

Only return `Disconnected`.

By the way,none of them prompted me to enter the password. I remember setting the password in `import webrepl_setup`.

wangshujun-tj, 2021-11-30:
I use thorny's webrepl connection. The same configuration is correct for esp32, but S2 will report an error. The initialization and setting seem to be normal

Wind-stormger, 2021-11-30:
> I use thonny webrepl connection. The same configuration is correct for esp32, but S2 will report an error. The initialization and setting seem to be normal

I tried thorny, too.

At first, I thought it was caused by the slight difference between tinyusb's CDC function and ordinary UART, but when I compiled the firmware, I redirected repl to output to UART0. Although repl was still output normally, the problem of webrepl was still not solved.

Maybe there are problems that I ignore or deeper problems that I can't solve.

unaiur, 2022-03-16:
There is a crypto bug that causes this issue. The easier way to solve it is disabling PSRAM in ESP32-S2 boards until Micropython is updated to a recent enough SDK.

dpgeorge, 2022-03-17:
This is now fixed, WebREPL works on ESP32-S2 when built using IDF 4.4.0 (the daily builds use this IDF version).

closed 2022-03-17 by dpgeorge