micropython/micropython #7729: Unexpected freezes for typical application.

no111u3, 2021-08-28:
When I start porting to new board #7728 I meet some strange freezes. Some console runs crash and firmware soft rebooted.
For long run typical graphic script I have freezes (demo app placed [gist])(https://gist.github.com/no111u3/632c22b7c285e88bb33c3bc3ddefee11) -- [video of run this app](https://youtu.be/JePjg_VeYW0)

no111u3, 2021-08-29:
Partially fixed by increase flash wait cycles.

JohnieBraaf, 2021-08-29:
Have you checked out this library. It seems to support double frame buffer and is implemented in C for the most part. It might perform better.

[LVGL Micropython binding](https://github.com/lvgl/lv_binding_micropython)

no111u3, 2021-08-29:
Thank for advice, but this app is partially stress test for micropython and I think we have some troubles with high load apps.

JohnieBraaf, 2021-08-29:
Can you identify the exact bottleneck, by measuring the time needed per method for example?
If you find slow perfoming methods/function you could try to optimize them with @micropython.native or @micropython.viper annotations.

dpgeorge, 2021-08-29:
I've been working on getting the stm32 port running on the H7B3 and had similar reliability issues, it would crash after a few seconds of running.  From memory the issue was VOS level and flash latency.  My working branch for this is here: https://github.com/dpgeorge/micropython/tree/stm32-h7b3-support