micropython/micropython #7731: Paste mode hangs ESP32 TTGO with FS mounted via mpremote

jdtsmith, 2021-08-31:
```
% mpremote connect /dev/tty.usbserial-XXXXXX mount . repl
Local directory . is mounted at /remote
Connected to MicroPython at /dev/tty.usbserial-XXXXXX
Use Ctrl-] to exit this shell
>
MicroPython v1.15-835-g1fcd6e97f-dirty on 2021-08-27; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
104976
>>> import lvgl as lv
>>> lv
<module 'lvgl'>
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== print(15)
=== 
15
>>> 
```

Hangs here, and reports:

```
>>> b''
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 445, in main
    do_repl(pyb, args)
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 332, in do_repl
    do_repl_main_loop(
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 259, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboardextended.py", line 613, in soft_reset_with_mount
    self.exec_("__mount(%s)" % (self.dev_out is not None))
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 465, in exec_
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 457, in exec_raw
    return self.follow(timeout, data_consumer)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 367, in follow
    raise PyboardError("timeout waiting for first EOF reception")
mpremote.pyboard.PyboardError: timeout waiting for first EOF reception

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 455, in main
    do_disconnect(pyb)
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 207, in do_disconnect
    pyb.enter_raw_repl(soft_reset=False)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 355, in enter_raw_repl
    raise PyboardError("could not enter raw repl")
mpremote.pyboard.PyboardError: could not enter raw repl
```

Paste-mode works fine when there is no FS mount.

peterhinch, 2021-08-31:
I can replicate this on the reference board, running today's firmware. The paste mode works, but then the ESP32 hangs before producing a dump similar to yours.
```python
[adminpete@capybara]: ~
$ mp
Local directory . is mounted at /remote
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell
>
MicroPython v1.16 on 2021-08-31; ESP32 module with ESP32
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== for n in range(10):
===     print('Hello')
=== 
Hello
Hello
Hello
Hello
Hello
Hello
Hello
Hello
Hello
Hello
>>> b''
Traceback (most recent call last):
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 452, in main
    do_repl(pyb, args)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 332, in do_repl
    do_repl_main_loop(
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 259, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboardextended.py", line 613, in soft_reset_with_mount
    self.exec_("__mount(%s)" % (self.dev_out is not None))
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 465, in exec_
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 457, in exec_raw
    return self.follow(timeout, data_consumer)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 367, in follow
    raise PyboardError("timeout waiting for first EOF reception")
mpremote.pyboard.PyboardError: timeout waiting for first EOF reception

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/adminpete/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 455, in main
    do_disconnect(pyb)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 207, in do_disconnect
    pyb.enter_raw_repl(soft_reset=False)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 355, in enter_raw_repl
    raise PyboardError("could not enter raw repl")
mpremote.pyboard.PyboardError: could not enter raw repl
[adminpete@capybara]: ~
```

peterhinch, 2021-08-31:
I have also reproduced this on a Pyboard D SF6W.

robert-hh, 2021-08-31:
I cannot replicate that on an ESP32, PYBD_SF6, RP2 of MIMXRT1010_EVK, with freshly built images, V1.16-259.
Terminal emulator used: picocom + board 
Connecting with rshell worked as well.  I did not test more boards & ports, but certainly can do that.

jdtsmith, 2021-08-31:
I have found this occurs at other times, for example on failed imports from /remote, so it seems to be more of a general issue than just paste mode. 

```
^[[C
b'\r\n\x18\x01'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 445, in main
    do_repl(pyb, args)
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 332, in do_repl
    do_repl_main_loop(
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 259, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboardextended.py", line 612, in soft_reset_with_mount
    self.exec_(fs_hook_code)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 465, in exec_
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 456, in exec_raw
    self.exec_raw_no_follow(command)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 423, in exec_raw_no_follow
    raise PyboardError("could not enter raw repl")
mpremote.pyboard.PyboardError: could not enter raw repl

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 455, in main
    do_disconnect(pyb)
  File "/usr/local/lib/python3.9/site-packages/mpremote/main.py", line 207, in do_disconnect
    pyb.enter_raw_repl(soft_reset=False)
  File "/usr/local/lib/python3.9/site-packages/mpremote/pyboard.py", line 355, in enter_raw_repl
    raise PyboardError("could not enter raw repl")
mpremote.pyboard.PyboardError: could not enter raw repl
```

jdtsmith, 2021-08-31:
May be related to #7732

peterhinch, 2021-12-22:
In my experience paste mode always crashes on a RP2 Pico if `mpremote` is issued with
```bash
$ mpremote mount .
```
It has caught me out numerous times and seems entirely repeatable.

It also leaves the REPL unresponsive on a Pyboard 1.1.

peterhinch, 2022-02-04:
Also happens on a Pyboard:
```python
$ mpremote mount .
Local directory . is mounted at /remote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] to exit this shell
>
MicroPython v1.18-56-g517e82eb6 on 2022-02-01; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
MPY: sync filesystems
MPY: soft reboot
MicroPython v1.18-56-g517e82eb6 on 2022-02-01; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def foo():
===     print('foo')
=== foo()
=== 
foo
>>> 
```
While this works, the Pyboard is in an unstable state with the REPL unresponsive. Then, after a while and with no user input, this happens:
```python
>>> b'\x18\x01\x07\x00\x00\x00/uio.py\x18\x01\x08\x00\x00\x00/uio.mpy\x18\x01\x08\x00\x00\x00/ustruct\x18\x01\x0b\x00\x00\x00/ustruct.py\x18\x01\x0c\x00\x00\x00/ustruct.mpy\x18\x01\x0c\x00\x00\x00/micropython\x18\x01\x0f\x00\x00\x00/micropython.py\x18\x01\x10\x00\x00\x00/micropython.mpy\x18\x01\x05\x00\x00\x00/usys\x18\x01\x08\x00\x00\x00/usys.py\x18\x01\t\x00\x00\x00/usys.mpy\x04\x04>'
Traceback (most recent call last):
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 452, in main
    do_repl(pyb, args)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 332, in do_repl
    do_repl_main_loop(
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 259, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboardextended.py", line 612, in soft_reset_with_mount
    self.exec_(fs_hook_code)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 467, in exec_
    raise PyboardError("exception", ret, ret_err)
mpremote.pyboard.PyboardError: ('exception', b'\x18\x01', b'\x00\x00\x00/uos\x18\x01\x07\x00\x00\x00/uos.py\x18\x01\x08\x00\x00\x00/uos.mpy\x18\x01')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/adminpete/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 455, in main
    do_disconnect(pyb)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 207, in do_disconnect
    pyb.enter_raw_repl(soft_reset=False)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 355, in enter_raw_repl
    raise PyboardError("could not enter raw repl")
mpremote.pyboard.PyboardError: could not enter raw repl
[adminpete@capybara]: ~
```

dpgeorge, 2022-02-04:
I can see the problem: ctrl-D is intercepted by the terminal handling loop (on the PC) and if `mpremote mount` is active then it sends ctrl-D to the board **followed by re-executing of the remote mounting code**.  It thinks that the ctrl-D did a soft reset, but it just ended the paste mode and now the PC is mounted twice on the board, leading to problems.

dpgeorge, 2022-02-04:
I think a decent fix would be for it to wait for `MPY: soft reboot` after passing through the ctrl-D, and only re-execute the mount code if it sees that string.

andrewleech, 2022-02-04:
Shouldn't mpremote be aware if the board should be in raw mode and skip the soft reboot hooks?

dpgeorge, 2022-02-04:
Well, ctrl-D can be sent in many different places, eg while entering text to delete the current character.  So mpremote would need to be very aware of the current state.  I'd say that's too difficult/flaky to do.

andrewleech, 2022-02-04:
Ah good point, yeah it's a wider problem than just raw mode. I guess a background task in mpremote that handles serial read and watches the stream for important markers could really help mpremote keep in sync with the device

dpgeorge, 2022-02-05:
See #8265 for a fix.

peterhinch, 2022-02-05:
Excellent! Tested on Pyboard 1.1 and RP2 Pico.

dpgeorge, 2022-02-06:
Thanks for testing!

dpgeorge, 2022-02-07:
Fixed by fecfd5269632e86f86c6b31a9bc2ec22876f8934

closed 2022-02-07 by dpgeorge