micropython/micropython #7733: ESP32-S2 + PSRAM

moefear85, 2021-08-31:
I read through the open issues.. I found alot on esp32-s2 support, some mention PSRAM, but none mention how to configure the build to get it working, or if that is necessary.

So.. I don't know where else to ask this... is it possible to compile MP for ESP32-S2 with (2MB) PSRAM enabled?

dpgeorge, 2021-09-01:
It might be enough to include `boards/sdkconfig.spiram` in the `set(SDKCONFIG_DEFAULTS ...)` part of your `mpconfigboard.cmake` file.

moefear85, 2021-09-02:
thx I'll give it a try... it leaves the question though, how would I know for sure after flashing that psram is enabled and being used?

currently cloning the latest master, to see if esp32s2 wifi works

dpgeorge, 2021-09-03:
> how would I know for sure after flashing that psram is enabled and being used?

Run:
```
>>> import micropython
>>> micropython.mem_info()
```
It will tell you how big the heap is.

closed 2023-11-06 by dpgeorge