micropython/micropython #7734: Serial keeps outputting after Ctrl+C

moefear85, 2021-08-31:
Sometimes I print debug info to Serial in a very tight loop, causing buffers to fill up. Thus after pressing Ctrl+C to end the running code, it can take up to even 20 seconds before the board responds. I'm not sure if micropython is simply caught up, but I assume it has long since processed and responded to the Ctrl+C and terminated the program, but continues to send all the "pent-up" data out over serial. The same might be happening on the PC side, so I'm not sure whether this should be a micropython issue or not.

But, is it possible, after micropython sees a Ctrl+C, that it immediately discard everything in the output buffer, since usually by the time someone presses Ctrl+C, they are no longer interested in the rest of the serial output.

dpgeorge, 2021-09-01:
The serial output buffer of the device (you didn't specify which one you are using though) is probably not very large, and MicroPython will stop printing as soon as it receives ctrl-C.  So the issue is likely on the PC side.

moefear85, 2021-09-04:
ah ok... I'll double check then thx. It happens really with any esp (8266,32,s2...).

dpgeorge, 2021-09-04:
If you can provide instructions to reproduce the issue then we could check if it's really a problem with the board (as opposed to the PC).

closed 2023-11-06 by dpgeorge