micropython/micropython #7736: Import modules from filesystem vs built-in

Lestat-GitHub, 2021-08-31:
Hi,

sorry if that has been already answered, but I did not find any information about my issue...

I have a custom firmware with custom modules (frozen and build into firmware).
For the example, let's say my module is "supermodule.py". This module is included into my firmware so I
can import it via "import supermodule".

Now if I want to test a new version and upload a copy of "supermodule.py" to the filesystem, the imported
module is still the one included in the firmware... I also tried to copy "supermodule.mpy" but the result is the
same.

Is there a way for force import from filesystem or does the built-in modules always have priority on import ?

NOTA: I'm using MicroPython 1.16 on an ESP8266

Thanks.

robert-hh, 2021-09-01:
This question has been many times asked and answered in the forum at https://forum.micropython.org/.
There is the list sys.path which defines the order in which places are searched for a module. The empty string represent the (virtual) path of the frozen modules. If you change the order in that list, you can import modules fro the file system in preference of the frozen modules. On my esp8266, the empty path is the first entry. so with:
sys.path.pop(0)
sys.path.append("")
you would make it the last entry. Another option would be to use the "mpremote run" to load an execute modules on-the-fly.

stinos, 2021-09-01:
Isn't this in the documentation as well somewhere? If not it should be probably :)

Lestat-GitHub, 2021-09-01:
Hi,
thanks for pointing me to the right solution !

None of my keyword helped me to find that... And as I'm more familiar with classical Python, I did not react when
I saw the empty string on sys.path list.

closed 2021-09-01 by Lestat-GitHub