micropython/micropython #7737: LinkError: build/cJSON.o: .data.rel non-empty

xijing-github, 2021-09-01:
What should I do? My code should be right

`GEN build/mpdata.config.h
CC cJSON.c
CC mpdata.c
CC modmpdata.c
LINK build/cJSON.o
LinkError: build/cJSON.o: .data.rel non-empty
make: *** [/home/xijing/micropython/py/dynruntime.mk:140: build/mpdata.native.mpy] Error 1`

project      [https://github.com/xijing-github/mpdata_map](url)

dpgeorge, 2021-09-04:
> LinkError: build/cJSON.o: .data.rel non-empty

It's not supported to have relocations in the data section.  `global_hooks` cannot be initialised, it must be like this:
```
static internal_hooks global_hooks;
```
and then an initilisation function added to populate this variable with its default values.

Apart from that, there are a lot of other symbols (like sprintf, sscanf, malloc) which won't be resolved by the linker.  I'd suggest trying USER_C_MODULES instead.

xijing-github, 2021-09-06:
> > LinkError: build/cJSON.o: .data.rel non-empty
> 
> It's not supported to have relocations in the data section. `global_hooks` cannot be initialised, it must be like this:
> 
> ```
> static internal_hooks global_hooks;
> ```I'll stick with C_USE_MODULES
> 
> and then an initilisation function added to populate this variable with its default values.
> 
> Apart from that, there are a lot of other symbols (like sprintf, sscanf, malloc) which won't be resolved by the linker. I'd suggest trying USER_C_MODULES instead.

Thank you, I also found these problems later, but there is no good solution.

closed 2021-09-06 by xijing-github